zephyrproject-rtos/zephyr #41911: pm_power_state_force returns false after first call

r2r0, 2022-01-18:
**Describe the bug**
I modified the zephyr/samples/boards/stm32/power_mgmt/stm32wb_ble/src/main.c file and I changed "pure" call to pm_power_state_force for sometihing to allow visually detect problems witch switching to shutdown:
```C
        if (!pm_power_state_force(0u, (struct pm_state_info){PM_STATE_SOFT_OFF, 0, 0})) {
                printk("Cannot force SHUTDOWN!\n");
        }
        k_sleep(K_MSEC(5));
        printk("Not in SHUTDOWN!\n");
```
And unfortunately above condition is met and also "Not in SHUTDOWN!" is printed.

**To Reproduce**
Steps to reproduce the behavior:
1. Change code in zephyr/samples/boards/stm32/power_mgmt/stm32wb_ble/src/main.c as described above
2. west build -b nucleo_wb55rg
3. west flash
4. Both messages from code above can be observed.

**Expected behavior**
No messages from modified code should be visible and MCU should switch to SHUTDOWN mode..

**Impact**
showstopper

**Logs and console output**
```
*** Booting Zephyr OS build v2.7.99-3296-ge8df8a579033  ***
Starting Beacon Demo
Bluetooth initialized
Beacon started, advertising as 80:E1:26:08:24:BC (public)
Beacon stopped
Beacon started
Device shutdown
Cannot force SHUTDOWN!
Not in SHUTDOWN!
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.13.1
 - Commit SHA e8df8a579033be238beb774e4909d637da649f3a

**Additional context**
I observed that `z_power_states_forced` in pm.c is not equal 0 when `pm_power_state_force` is called **first** time.

ceolin, 2022-01-18:
That is really weird, this function only returns `false` if the cpu bit in `z_power_states_forced` is 1 and this should be zeroed initialized. Can you try to following change and see what happens ?
```
@@ -29 +29 @@ static struct pm_state_info z_power_states[CONFIG_MP_NUM_CPUS];
-static ATOMIC_DEFINE(z_power_states_forced, CONFIG_MP_NUM_CPUS);
+static ATOMIC_DEFINE(z_power_states_forced, CONFIG_MP_NUM_CPUS) = { 0 };
```

r2r0, 2022-01-19:
Unfortunately it changes nothing.
I suspect that problem can be caused by the fact that `z_power_states_forced` flag is cleared **after** scheduler is unlocked (in `pm_system_suspend`).
And then `pm_system_suspend` is called before main (issue #41839) then after main is scheduled the `z_power_states_forced` flag is still set and it prevents proper `pm_power_state_force` functionality.
Even if there was no sleep before main then such control flow (scheduler unlock -> flag clearing) is IMHO problematic.

ceolin, 2022-01-19:
Ohh, I see. I have overlooked the messages, the first call works. Will look it better.

ceolin, 2022-01-28:
Yep, I can confirm, if we clear out the flags with the scheduler locked the problem seems to go away. The assessment is correct, the main thread is scheduled after the system resumes and before the flags being cleared out.

PR #42274 (2022-02-01): pm: Fix state_force functionality
The "state forced" flag has to be cleared with the scheduler locked,
otherwise the idle thread can be scheduled out without clear it and
the next call to force a state will fail.

With the forced flag been cleared with the kernel locked in the
suspend path, we need to clear it out when the suspend process fails.

This is mow done before jump to the end label because in the
successful path the flag is already cleared.

Fixes #41911
commit fdf74658 (2022-02-01): pm: Fix multithread issue with force state

The "state forced" flag has to be cleared with the scheduler locked,
otherwise the idle thread can be scheduled out without clear it and
the next call to force a state will fail.

Fixes #41911
closed 2022-02-01 by nashif