zephyrproject-rtos/zephyr #41925: conflict of v2.7-branch with main:

mtahirbutt, 2022-01-18:
Hi mates, 
              I am trying to pull v2.7-branch but it shows conflicts with many files and resultantly pull is not successful. I downloaded fresh repo from zephyr and it is definitely 2.7.99 (development version). But I want to work with v2.7. 

Somebody can help me please comment to sort out this issue to work with v2.7.

regards,

Tahir

lsangild, 2022-01-19:
You cannot pull to an existing branch, this is akin to merging the two branches. What you want to do is to switch to another branch, which can be done with the `git checkout` feature.

Delete your downloaded repository folder and clone a new one.

You can either checkout to the 2.7 branch, which is still in development.

`git checkout v2.7-branch`

Or if you want a static 2.7 you should checkout to the 2.7.0 tag (a tag is a shortcut to a certain commit).

`git checkout v2.7.0`

mtahirbutt, 2022-01-19:
> thank you very much for your kind advice

closed 2022-01-19 by henrikbrixandersen