zephyrproject-rtos/zephyr #41933: updatehub  metadata size 0

youssefchei-xite, 2022-01-18:
**Describe the bug**
I'm testing updatehub on nRF53dk and BG95 as a modem.
I'm using updatehub cloud without DTLS.

- I flashed the updatehub example and it works and connected to **updatehub cloud**.
- Then I build a hello sample, signed then pushed to **updatehub cloud** and finally create a rollout.
 
When I tried to start the update using **updatehub run** command I got 

```
uart:~$ updatehub run
Starting UpdateHub run...
Invalid response
[00:01:07.832,550] <inf> updatehub: Probe metadata received
[00:01:07.832,855] <err> updatehub: Could not parse json
```
I tried different configurations to generate the signed image and pushed it to **updatehub cloud**

What is this config option supposed to be, and how can I get it to work? Am I missing another config option?

**To Reproduce**
Steps to reproduce the behavior:

1. build the updatehub sample

> west build --board=nrf5340dk_nrf5340_cpuapp -d build zephyr/samples/subsys/mgmt/updatehub -- \
> -DOVERLAY_CONFIG="overlay-modem.conf"

2. flash the binaries
> west flash -d build
 
output :

```
[00:00:48.009,399] <inf> main: starting  update hup process ......
[00:00:48.009,399] <inf> main: Starting UpdateHub manual mode
[00:00:49.179,443] <inf> updatehub: No update available
[00:00:49.179,534] <inf> main: No update found
```

3. build the hello world sample 
> west build --board=nrf5340dk_nrf5340_cpuapp -d build_hello zephyr/samples/hello_world -- \
> -- -DCONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_UPDATEHUB=y
4. sign the binary 
> west sign -t imgtool -d build_hello -- --version 5.5.5 --pad --header-size 0x200 --align 8 \           
>     --key  bootloader/mcuboot/root-rsa-2048.pem

I got an error:
```
=== image configuration:
partition offset: 65536 (0x10000)
partition size: 262144 (0x40000)
rom start offset: 0 (0x0)
=== signing binaries
unsigned bin: build_hello/zephyr/zephyr.bin
signed bin:   build_hello/zephyr/zephyr.signed.bin
Usage: imgtool sign [OPTIONS] INFILE OUTFILE
Try 'imgtool sign -h' for help.
Error: Header padding was not requested and image does not start with zeros  
```
5. build again the hello world sample with CONFIG_ROM_START_OFFSET=0x200 
> west build --board=nrf5340dk_nrf5340_cpuapp -d build_hello zephyr/samples/hello_world -- \
> -- -DCONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_UPDATEHUB=y -CONFIG_ROM_START_OFFSET=0x200 

 6. sign the binary 
> west sign -t imgtool -d build_hello -- --version 5.5.5 --pad --header-size 0x200 --align 8 \           
>     --key  bootloader/mcuboot/root-rsa-2048.pem
output files signed 

```
=== image configuration:
partition offset: 65536 (0x10000)
partition size: 262144 (0x40000)
rom start offset: 512 (0x200)
=== signing binaries
unsigned bin: build_hello/zephyr/zephyr.bin
signed bin:   build_hello/zephyr/zephyr.signed.bin
unsigned hex: build_hello/zephyr/zephyr.hex
signed hex:   build_hello/zephyr/zephyr.signed.hex
```

7. Create and push the update package 5.5.5 to **updatehub cloud**  using https://docs.updatehub.io/zephyr-project/zephyr-project-guide/ , steps: 8-9 then create rollout using step 11

8. force the update using **updatehub run** command

output:
```
uart:~$ updatehub run
Starting UpdateHub run...
Invalid response
[00:01:07.832,550] <inf> updatehub: Probe metadata received
[00:01:07.832,855] <err> updatehub: Could not parse json
```
I have checked the size of the metadata received and it's always 0

**Expected behavior**
After running **updatehub run**, the download of the firmware complete

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain :Zephyr SDK
 - Version used:v2.6.99-ncs1

nandojve, 2022-01-19:
Hi @youssefchei-xite ,

I think you are doing everything OK. However, you are using an intermediate version, not an official ZephyrRTOS release.
I'll ask you to check if this #37854 fix your problem. Please, test and follow up.

youssefchei-xite, 2022-01-21:
Hi @nandojve 
I'm using the version provided by **nRF Connect SDK**  v1.7.0 [https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/gs_installing.html].
After integrating the modification from commit #37854, it works and I could do the OTA update.

Now, I want to test updatehub with DTLS, so, I have enabled the DTLS using **overlay-modem.conf** and **overlay-dtls.conf**. This Time I got TLS error

> Network c[00:00:30.047,393] <inf> main: starting  update hup process ......
> uart:~$
> Starting UpdateHub man[00:02:33.438,079] <err> net_sock_tls: TLS handshake error: -6800
> [00:02:33.446,990] <err> updatehub: Could not send request

Is there a Certificate or passkey to make the updatebup over DTLS work?

youssefchei-xite, 2022-01-25:
Hi @nandojve
I'm using the version provided by nRF Connect SDK v1.7.0 [https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/gs_installing.html].
After integrating the modification from commit #37854, it works and I could do the OTA update.

closed 2022-01-25 by youssefchei-xite