zephyrproject-rtos/zephyr #41942: k_delayable_work being used as k_work in work's handler

ycsin, 2022-01-19:
The `k_delayable_work` being passed into the work handler as `struct k_work *work` are being used directly just like a `work` and are passed into `CONTAINER_OF` to derive the pointer to the parent struct.

I imagine that the worst case scenario being the code modify some unintended memory addresses and cause corruptions, but I'm not sure.

See https://github.com/ycsin/zephyr/commit/47c0dd506549bbff8bc67b5ff37288ac4faefd60, this commit is done by searching for `CONTAINER_OF(work`, and manually checking them, so there might be human error and if the handler k_work arg isn't named as the conventional `work`, then it wont appear in the search result.

ycsin, 2022-01-19:
@andyross @pabigot

ycsin, 2022-01-19:
@andyross seems to be pretty common in Zephyr that the `k_work_delayable` to be used as a timer, some of them are being named directly as a timer, i.e. `struct k_work_delayable timer`

pabigot, 2022-01-19:
Yes, according to the [documentation](https://docs.zephyrproject.org/latest/reference/kernel/threads/workqueue.html?highlight=delayable%20work#delayable-work) the change you're proposing is correct, though technically unnecessary as long as `work` is the first item in the `k_work_delayable` structure.  Other [existing code](https://github.com/zephyrproject-rtos/zephyr/blob/7dac931e3662eb666cdabc3bf1759a7484249730/subsys/bluetooth/audio/csis.c#L481-L483) uses the more precise sequence.

The conversions were done (by coccinelle) which could not determine that this "misuse" was pre-existing in code that used the previous delayable work solution.

For code quality purposes fixing this seems wise.

closed 2022-02-02 by nashif