zephyrproject-rtos/zephyr #41943: Intel_adsp_cavs15:   all the test cases run failed when running them by twister

enjiamai, 2022-01-19:
**Describe the bug**
The twister cannot be used to testing intel_adsp_cavs15 after this commit 8590f67f698e2492a6757239da13a945f88a6646
But when we manually run it by west, it will pass the test.

 - What target platform are you using? 
 intel_adsp_cavs15 (on up_squared board)

 - What have you tried to diagnose or workaround this issue?
I found the test case failed in twister because when the adsplog.py start to catch the message then trigger error,  
but at this moment the building and signing for zephyr.ri is not completed yet.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. run: 
twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/pipe/pipe --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/rimage/config,/home/ztest/work/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history" -vv -j 1 

3. you can see the screen has no output and hang in some point . See error log below please.

**Expected behavior**
The test case will pass and see the output in your screen.

**Impact**
The daily testing pass rate became zero passrate.

**Logs and console output**
```
ztest@zserverUP2:~/work/zephyrproject/zephyr$ twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/pipe/pipe --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/rimage/config,/home/ztest/work/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history" -vv
Renaming output directory to /home/ztest/work/zephyrproject/zephyr/twister-out.15
INFO    - Zephyr version: v2.7.99-3293-g93b0ea978293
INFO    - JOBS: 4
DEBUG   - Reading test case configuration files under /home/ztest/work/zephyrproject/zephyr/tests/kernel/pipe/pipe...
DEBUG   - Found possible test case in /home/ztest/work/zephyrproject/zephyr/tests/kernel/pipe/pipe
DEBUG   - Reading platform configuration files under /home/ztest/work/zephyrproject/zephyr/boards...
DEBUG   - Reading platform configuration files under /home/ztest/work/zephyrproject/zephyr/scripts/pylib/twister/boards...
DEBUG   - Running cmake script /home/ztest/work/zephyrproject/zephyr/cmake/verify-toolchain.cmake
DEBUG   - Calling cmake with arguments: ['-DFORMAT=json', '-P', PosixPath('/home/ztest/work/zephyrproject/zephyr/cmake/verify-toolchain.cmake')]
DEBUG   - Finished running  /home/ztest/work/zephyrproject/zephyr/cmake/verify-toolchain.cmake
INFO    - Using 'zephyr' toolchain.
DEBUG   - platform filter: ['intel_adsp_cavs15']
DEBUG   -     arch_filter: None
DEBUG   -      tag_filter: None
DEBUG   -     exclude_tag: None
INFO    - Building initial testcase list...
INFO    - 1 test scenarios (1 configurations) selected, 0 configurations discarded due to filters.

Device testing on:

| Platform          | ID   | Serial device   |
|-------------------|------|-----------------|
| intel_adsp_cavs15 |      |                 |

INFO    - Adding tasks to the queue...
DEBUG   - adding intel_adsp_cavs15/tests/kernel/pipe/pipe/kernel.pipe
INFO    - Added initial list of jobs to queue
DEBUG   - Launch process 0
DEBUG   - Launch process 1
DEBUG   - Launch process 2
DEBUG   - Launch process 3
DEBUG   - Running cmake on /home/ztest/work/zephyrproject/zephyr/tests/kernel/pipe/pipe for intel_adsp_cavs15
DEBUG   - Calling cmake with arguments: ['-B/home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/pipe/pipe/kernel.pipe', '-S/home/ztest/work/zephyrproject/zephyr/tests/kernel/pipe/pipe', '-DEXTRA_CFLAGS=-Werror', '-DEXTRA_AFLAGS=-Werror -Wa,--fatal-warnings', '-DEXTRA_LDFLAGS=-Wl,--fatal-warnings', '-DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror', '-GUnix Makefiles', '-DBOARD=intel_adsp_cavs15']
DEBUG   - Finished building /home/ztest/work/zephyrproject/zephyr/tests/kernel/pipe/pipe for intel_adsp_cavs15
DEBUG   - build test: intel_adsp_cavs15/tests/kernel/pipe/pipe/kernel.pipe
DEBUG   - Building /home/ztest/work/zephyrproject/zephyr/tests/kernel/pipe/pipe for intel_adsp_cavs15
DEBUG   - run test: intel_adsp_cavs15/tests/kernel/pipe/pipe/kernel.pipe
DEBUG   - Using serial device /dev/pts/0 @ 115200 baud
DEBUG   - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/pipe/pipe/kernel.pipe', '--', '/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh', '/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem', '/home/ztest/work/rimage/config', '/home/ztest/work/rimage/build/rimage']

```
And there is nothing output in the twister-out/intel_adsp_cavs15/tests/kernel/pipe/pipe/kernel.pipe/handler.log

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK, 0.13.2
 - Commit SHA or Version : 8cb5bf152686ad9060162ce5b0215384536e9585

**Additional context**
 The test will pass when we manually run following command:
1. west build -b intel_adsp_cavs15 -p always tests/kernel/mutex/mutex_api
2. west sign -d build -t rimage -p /home/ztest/work/rimage/build/rimage -D /home/ztest/work/rimage/config -- -k /home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem --no-manifest
3. boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py -f build/zephyr/zephyr.ri
4. boards/xtensa/intel_adsp_cavs15/tools/adsplog.py

andyross, 2022-01-19:
That commit changed the output protocol.  You can't use an older script with a newer firmware binary.  Are you sure that "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py" (the script twister is using) and "boards/xtensa/intel_adsp_cavs15/tools/adsplog.py" (the script you used with west) are identical?

enjiamai, 2022-01-19:
Yes, @andyross. They are identical, the new adsplog.py using new winstream protocol to get log from ADSP. I use that new one.
And I found it the twister works by doing following trial:

```
diff --git a/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py b/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
index 61568c811c..10ad1c250d 100755
--- a/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
+++ b/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
@@ -29,6 +29,9 @@ WIN_IDX = 3
 WIN_SIZE = 0x20000
 LOG_OFFSET = WIN_OFFSET + WIN_IDX * WIN_SIZE

+# just a trial workaround, wait 15 sec for the image built and signed.
+time.sleep(15)
+
 mem = None
 sys_devices = "/sys/bus/pci/devices"

@@ -102,3 +105,4 @@ while True:
     (last_seq, output) = winstream_read(last_seq)
     if output:
         sys.stdout.write(output)
+        sys.stdout.flush()

```

enjiamai, 2022-01-19:
Hi @andyross , by the way, may I ask you one question. When I run it manually by west, the test pass, but I saw this message, it that matter?  
```
File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/ipc.py", line 242, in read_fw_ready_ext
    raise RuntimeError("Not Implemented: ext_hdr.type != 1")
```
here is the full message:
```
Reset DSP...
Firmware Status Register (0xFFFFFFFF)
   Boot:   0xFFFFFF (UNKNOWN)
   Wait:   0x0F (UNKNOWN)
   Module: 0x07
   Error:  0x01
   IPC CMD  : 0xFFFFFFFF
   IPC LEN  : 0xFFFFFFFF
Booting up DSP...
Firmware Status Register (0x05000001)
   Boot:   0x000001 (INIT_DONE)
   Wait:   0x05 (DMA_BUFFER_FULL)
   Module: 0x00
   Error:  0x00
Downloading firmware...
Start firmware downloading...
Checking firmware status...
Firmware Status Register (0x00000005)
   Boot:   0x000005 (FW_ENTERED)
   Wait:   0x00 (UNKNOWN)
   Module: 0x00
   Error:  0x00
Firmware download completed successfully
Reading IPC FwReady Message...
IPC Firmware Ready Message: (0x00000000) (0x00000000)
   DSP box offset:     0 (0x0000)
   Host box offset:    0 (0x0000)
   DSP box size:       0 (0x0000)
   Host box size:      0 (0x0000)

Firmware Version:
   Major version:      0 (0x0000)
   Minor version:      0 (0x0000)
   Micro number:       0 (0x0000)
   Build number:       0 (0x0000)
   Date:
   Time:
   Tag:
   Abi version:     0 (0x0000)

Flags:   0x00000000

IPC Firmware Ready Extended Message: (0x00000000) (0x00000000)
Message Type:          0 (IPC_EXT_DMA_BUFFER)

Number of Windows:     0

Traceback (most recent call last):
  File "boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py", line 81, in <module>
    main()
  File "boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py", line 71, in main
    fw_loader.ipc.read_fw_ready()
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/ipc.py", line 233, in read_fw_ready
    self.read_fw_ready_ext()
  File "/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/lib/ipc.py", line 242, in read_fw_ready_ext
    raise RuntimeError("Not Implemented: ext_hdr.type != 1")
RuntimeError: Not Implemented: ext_hdr.type != 1
 PASS - test_pipe_get_timeout in 1.601 seconds
===================================================================
START - test_pipe_get_invalid_size
 PASS - test_pipe_get_invalid_size in 0.1 seconds
===================================================================
START - test_pipe_get_min_xfer
 PASS - test_pipe_get_min_xfer in 0.1 seconds
===================================================================
START - test_pipe_put_min_xfer
 PASS - test_pipe_put_min_xfer in 0.1 seconds
===================================================================
START - test_pipe_define_at_runtime
 PASS - test_pipe_define_at_runtime in 0.21 seconds
===================================================================
START - test_pipe_flush
 PASS - test_pipe_flush in 0.301 seconds
===================================================================
Test suite test_pipe succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

andyross, 2022-01-19:
Gah.  **Please** don't use fw_loader.py.  That tooling is ancient.  The official loaders have been the cavs-fw-*.py scripts for months now, which have now been unified (in #41828) as a "cavstool.py" that does both loading and trace output.

Load with cavs-fw-*.py.  Read logs with adsplog.py.  Use both from the same directory in which you are building your firmware.  If the directions for test integration don't match what you are doing, submit matches to rewrite or remove them.

enjiamai, 2022-01-20:
It really helps. @andyross . This was not enabled by me so I were not aware of that. According to your suggestion I try to trim the twister running from fw_loader.py to cavs-fw-*.py with minor change in adsplog.py. Now I can get 96.77% test pass rate on Up_squared ADSP. Next step I will try the new tool "cavstool.py" with twister. Thanks for the comment.

So this cavstool.py is aim to fit for all ADSP platform, include TGL. WHL, is my understanding correct?

andyross, 2022-01-20:
Yes, cavstool will be a single tool for all hardware use cases.  Once it merges I promise to remove all the old scripts and rewrite the docs.

marc-hb, 2022-01-20:
> Once it merges I promise to remove all the old scripts and rewrite the docs.

In the mean time adding a one-line warning "this tool is deprecated, use this other one" would likely save a lot of time. @enjiamai could you shared your new and hard learned knowledge and add these warnings?

cc: @keqiaozhang

closed 2022-01-21 by nashif

marc-hb, 2022-01-21:
@nashif or @andyross  or @teburd  can you please share a pointer to some PR? I see only links to items still opened or closed without merge above.

andyross, 2022-01-21:
Yeah, things were moving pretty quickly today.  https://github.com/zephyrproject-rtos/zephyr/pull/42020 is the PR with the scripting changes.  Also one merged with some test-specific fixups, and @teburd had one that adjusted the global filtering rules for the cavs platforms to expand the test set.

teburd, 2022-01-21:
#42055 I expanded the tests being run on cavs25 and fixed a minor test filtering issue