zephyrproject-rtos/zephyr #41947: lpcxpresso55s16 SPI hardware chip select not working

metratec-mkoehler, 2022-01-19:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

henrikbrixandersen, 2022-01-19:
Please edit the template text in this issue to provide more information than just the title. 

E.g. how did you reproduce this? How did you conclude it isn't working, etc.

metratec-mkoehler, 2022-01-19:
Sorry, I did add text and am unsure why it / I switched back to the default description. Will edit soon.

metratec-mkoehler, 2022-01-19:
Done. Have not found an easy to use example but will build a main() code to test if needed.

dleach02, 2022-02-01:
@metratec-mkoehler are you observing this problem on the lpcxpresso55s16 EVK board? I run the tests/drivers/spi/spi_loopback where I had to modify the lpc55s16 project file to disable CONFIG_SPI_ASYNC to get the test to complete. With that setup I observe the CS working.

[image]

dleach02, 2022-03-07:
Closed due to no response from @metratec-mkoehler. Please reopen if you have additional information