zephyrproject-rtos/zephyr #41953: drivers: counter: mcux_ctimer: config used as non-const

gmarull, 2022-01-19:
**Describe the bug**

These drivers seem to treat device configuration as non-const. For example:

https://github.com/zephyrproject-rtos/zephyr/blob/7dac931e3662eb666cdabc3bf1759a7484249730/drivers/counter/counter_mcux_ctimer.c#L186-L190

**To Reproduce**

1. Don't cast config to non-const (as it should be):
```
const struct mcux_lpc_ctimer_config *config = dev->config;
```
2. Build a board that uses the driver

**Expected behavior**

Device driver should use `data` to store non-ROMable data.

**Impact**

Impacts https://github.com/zephyrproject-rtos/zephyr/pull/41918

**Logs and console output**

N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.13.1
 - Commit SHA or Version used e8df8a579033be238beb774e4909d637da649f3a

**Additional context**

N/A

dleach02, 2022-01-25:
yep... that is a bug.

dleach02, 2022-01-25:
This seems like something static analysis should catch. Casting a const value into a non-const pointer..

PR #42133 (2022-01-26): drivers: counter: mcux_ctimer: fix config used as non-const
The initialization code was updating the freq field in the const
declared config block at runtime because the frequency is not
known at compile time. Add a get_freq() api call to handle any
runtime requests for frequency.

fixes #41953
commit 8ab9ac26 (2022-01-26): drivers: counter: mcux_ctimer: fix config used as non-const

The initialization code was updating the freq field in the const
declared config block at runtime because the frequency is not
known at compile time. Add a get_freq() api call to handle any
runtime requests for frequency.

fixes #41953
closed 2022-01-26 by dleach02