zephyrproject-rtos/zephyr #41993: Intel_adsp_cavs18: test cases can not get complete log

LixinGuoX, 2022-01-20:
**Describe the bug**
For test cases, binary image can be downloaded to ADSP, and there is some output can be caught by adsplog.py, but the log is not complete.

**To Reproduce**
Steps to reproduce the behavior:
use twister command:
`twister -p intel_adsp_cavs18 --device-testing -T tests/ --west-flash="/home/ztest/zephyrproject/UP_WHL/tools/flash18.sh" --device-serial-pty="/home/ztest/zephyrproject/UP_WHL/tools/adsplog.sh,--no-history" -j 1`
here is the flash18.sh and adsplog.sh:
```
flash18.sh
#!/bin/sh

BUILD=$1
FIRMWARE=${BUILD}/zephyr/zephyr.ri
FLASHER1=/home/ztest/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh
FLASHER=/home/ztest/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/cavs-fw-v15.py
KEY=/home/ztest/zephyrproject/modules/audio/sof/keys/otc_private_key.pem
CONFIG=/home/ztest/zephyrproject/modules/audio/sof/rimage/config
DIR=/home/ztest/zephyrproject/modules/audio/sof/rimage/build/rimage

echo "Signing with key " $key ${BUILD}
west sign -d ${BUILD} -t rimage -p ${DIR} -D ${CONFIG} -- -k ${KEY} --no-manifest

echo ${FLASHER} ${FIRMWARE}
timeout 60 ssh root@localhost ${FLASHER} ${FIRMWARE}

ssh root@localhost pkill -f -9 cavs-fw-v15.py || true

```
```
adsplog.sh
#!/bin/sh

ADSPLOG=/home/ztest/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py

ssh root@localhost pkill -f -9 adsplog.py || true

timeout 300 ssh root@localhost python3 -u ${ADSPLOG} $1
```

**Impact**
The log indicates a problem, it seems that it is too fast for adsplog.py to get the log.

**Logs and console output**
example log as below:
```
DEBUG   - Running cmake on /home/ztest/zephyrproject/zephyr/tests/kernel/mutex/mutex_api for intel_adsp_cavs18
DEBUG   - Calling cmake with arguments: ['-B/home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs18/tests/kernel/mutex/mutex_api/kernel.mutex', '-S/homgs', '-DEXTRA_LDFLAGS=-Wl,--fatal-warnings', '-DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror', '-GUnix Makefiles', '-DBOARD=intel_adsp_cavs18']
DEBUG   - Finished building /home/ztest/zephyrproject/zephyr/tests/kernel/mutex/mutex_api for intel_adsp_cavs18
DEBUG   - build test: intel_adsp_cavs18/tests/kernel/mutex/mutex_api/kernel.mutex
DEBUG   - Building /home/ztest/zephyrproject/zephyr/tests/kernel/mutex/mutex_api for intel_adsp_cavs18
DEBUG   - run test: intel_adsp_cavs18/tests/kernel/mutex/mutex_api/kernel.mutex
DEBUG   - Using serial device /dev/pts/49 @ 115200 baud
DEBUG   - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs18/tests/kernel/mutex/mutex_ap
DEBUG   - DEVICE: Killed
DEBUG   - DEVICE: access resource from main thread
DEBUG   - DEVICE: PASS - test_mutex_reent_lock_forever in 1.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mutex_reent_lock_no_wait
DEBUG   - DEVICE: access resource from main thread
DEBUG   - DEVICE: bypass locked resource from spawn thread
DEBUG   - -- west flash: using runner misc-flasher
Signing with key  /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs18/tests/kernel/mutex/mutex_api/kernel.mutex
Signing for SOC target cnl using cnl.toml
Signing with tool /home/ztest/zephyrproject/modules/audio/sof/rimage/build/rimage
/home/ztest/zephyrproject/modules/audio/sof/rimage/build/rimage -k /home/ztest/zephyrproject/modules/audio/sof/keys/otc_private_key.pem -o /home/ztest/zephyrpect/modules/audio/sof/rimage/config/cnl.toml -i 3 /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs18/tests/kernel/mutex/mutex_api/kernel.mutex/zepn.mod

Module Reading /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs18/tests/kernel/mutex/mutex_api/kernel.mutex/zephyr/boot.mod
info: ignore .bss section for bootloader module
  Found 6 sections, listing valid sections......
[... 113 lines cut ...]
DEBUG   - DEVICE: access resource from main thread
DEBUG   - DEVICE: bypass locked resource from spawn thread
DEBUG   - DEVICE: PASS - test_mutex_reent_lock_timeout_fail in 2.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mutex_reent_lock_timeout_pass
DEBUG   - DEVICE: access resource from main thread
DEBUG   - DEVICE: access resource from spawn thread
DEBUG   - DEVICE: access resource from main thread
DEBUG   - DEVICE: bypass locked resource from spawn thread
DEBUG   - DEVICE: PASS - test_mutex_reent_lock_timeout_pass in 2.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mutex_recursive
DEBUG   - DEVICE: PASS - test_mutex_recursive in 0.2 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mutex_priority_inheritance
DEBUG   - DEVICE: PASS - test_mutex_priority_inheritance in 6.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: Test suite mutex_api succeeded
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: PROJECT EXECUTION SUCCESSFUL
DEBUG   - Process /home/ztest/zephyrproject/UP_WHL/tools/adsplog.sh,--no-history terminated outs: None errs None
DEBUG   - run status: intel_adsp_cavs18/tests/kernel/mutex/mutex_api/kernel.mutex passed
INFO    - 40/87 intel_adsp_cavs18         tests/kernel/mutex/mutex_api/kernel.mutex          PASSED (device 13.215s)

```

teburd, 2022-01-21:
@LixinGuoX can you please reverify with the updated cavstwist.sh script @andyross wrote as I think that would fix this particular issue

LixinGuoX, 2022-01-24:
Yes, this issue is fixed by cavstwist.sh, so close this issue.

closed 2022-01-24 by LixinGuoX

enjiamai, 2022-01-24:
Verified this is fixed by #42020