zephyrproject-rtos/zephyr #41995: tracing: risvc: Missing invoking the sys_trace_isr_exit()

WealianLiao, 2022-01-20:
**Describe the bug**
RISCV core doesn't invoke the sys_trace_isr_exit() for tracing system.

**To Reproduce**
Apply PR:41863 & 
```
./scripts/twister -T ./samples/subsys/tracing/ -p qemu_riscv32
```
or
```
./scripts/twister -T ./samples/subsys/tracing/ -p qemu_riscv64
```

**Expected behavior**
In the report log, the ISR counter should not increase.

**Impact**
Can't trace ISR leave for RISCV.

WealianLiao, 2022-01-20:
This probably works but I don't have real the RISCV platform to check.
```
diff --git a/arch/riscv/core/isr.S b/arch/riscv/core/isr.S
index 1ff0144315..e09d3da1aa 100644
--- a/arch/riscv/core/isr.S
+++ b/arch/riscv/core/isr.S
@@ -235,6 +235,7 @@ GTEXT(z_thread_mark_switched_in)
 GTEXT(z_thread_mark_switched_out)
 #ifdef CONFIG_TRACING
 GTEXT(sys_trace_isr_enter)
+GTEXT(sys_trace_isr_exit)
 #endif
 #endif
 
@@ -735,6 +736,10 @@ call_irq:
        /* Call ISR function */
        jalr ra, t1
 
+#ifdef CONFIG_TRACING_ISR
+       call sys_trace_isr_exit
+#endif
+
 on_thread_stack:
        /* Get reference to _kernel */
        la t1, _kernel
```

edersondisouza, 2022-02-09:
Hi @WealianLiao,

The patch you pasted seems fine and it works on QEMU - I think you should just send it as a PR so it can be reviewed and merged!

edersondisouza, 2022-02-23:
@WealianLiao - gentle reminder about sending your patch as a PR - if you have already done so, please link it here.

closed 2022-05-09 by github-actions[bot]