zephyrproject-rtos/zephyr #42010: intel_adsp_cavs18: Test cases failed on SMP related test cases (when CONFIG_MP_NUM_CPUS > 1)

enjiamai, 2022-01-20:
**Describe the bug**
Many test suites failed in intel_adsp_cavs18 hardware board, this issue is likely to be SMP related:

tests/kernel/smp/ (no output)
tests/kernel/spinlock/ (no output)
tests/kernel/mp/ (no output)
tests/kernel/workq/work/ (no output, but with -DCONFIG_MP_NUM_CPUS=1 only test_smp_simple_queue, test_1cpu_sync_queue, test_smp_running_cancel)

And bellow test suites failed without any output log, but they can pass with -DCONFIG_MP_NUM_CPUS=1:

tests/kernel/profiling/profiling_api/ 
tests/kernel/mem_protect/sys_sem/
tests/kernel/msgq/msgq_api/
tests/kernel/msgq/msgq_usage/ 
tests/kernel/queue/
tests/kernel/obj_tracking/ 
tests/kernel/events/event_api/ 
tests/kernel/fatal/message_capture/
tests/kernel/mbox/mbox_usage/ 
tests/kernel/threads/thread_stack/
tests/kernel/lifo/lifo_api/ 
tests/kernel/mutex/sys_mutex/ 
tests/kernel/mem_slab/mslab_concept/
tests/kernel/fifo/fifo_api/ 
tests/kernel/timer/timer_monotonic/ 
tests/kernel/mem_heap/k_heap_api/ 
tests/kernel/mem_heap/mheap_api_concept/ 
tests/kernel/condvar/condvar_api/ 
tests/kernel/semaphore/smaphore/
tests/kernel/device/

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b intel_adsp_cavs18 -p always -t tests/kernel/smp/
2. west sign -d build -t rimage -p /home/ztest/zephyrproject/modules/audio/sof/rimage/build/rimage -D /home/ztest/zephyrproject/modules/audio/sof/rimage/config -- -k /home/ztest/zephyrproject/modules/audio/sof/keys/otc_private_key.pem --no-manifest
3. ssh root@localhost "boards/xtensa/intel_adsp_cavs15/tools/cavs-fw-v15.py build/zephyr/zephyr.ri"
4. python3 -u ${ZEPHYR_BASE}/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
5. You can see there is no output from ADSP.

**Expected behavior**
There should be testing messaged output. No matter it passed or failed

**Impact**
The whole kernel test pass rate down to 50%.

**Logs and console output**
```
Firmware completing manifest v1.8
 meta: completing ADSP manifest
 meta: limit is 0x16b80
 auth: completing authentication manifest
 cse: ri_css_v1_8_hdr_create completing CSS manifest
 css: set build date to 2022:00:20
 cse: completing CSE V1.8 manifest
Firmware file size 0x1b000 page count 21
 rimage_read_key: read key '/home/ztest/zephyrproject/modules/audio/sof/keys/otc_private_key.pem'
 pkcs: RSA private key is valid.
 pkcs: digest for manifest is a237faea7ee4b0e8a2efff1091e9a3983f44cab3013548d78106613a9d187ed0
Firmware manifest and signing completed !
INFO:cavs-fw:cAVS firmware loader v15
INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
INFO:cavs-fw:Detected cAVS 1.8+ hardware
INFO:cavs-fw:Obtained the physical address of the mapped huge page.
INFO:cavs-fw:Mapped 2M huge page at 0x15a200000 to contain 86016 bytes of firmware
INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
INFO:cavs-fw:ADSPCS = 0xf0f0000

(hang up and no output here .....)
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain: Zephyr SDK 0.13.2
 - Commit SHA or Version used: f08c44def5fc

**Additional context**

andyross, 2022-01-20:
I don't have this hardware currently, so I'm going to take myself off as assignee while we find someone else to look at this.  Most likely this is a kconfig mistake or accidental regression with recent work.  The MP mechanisms here are almost the same as 1.5 and 2.5.  And IIRC SOF has 1.8 zephyr multiprocessor setups that don't seem to be failing.

andyross, 2022-01-22:
Indeed, secondary core startup wasn't working (but amusingly in a way that SOF was insulated from by an existing workaround, so they never noticed).  Fix up at https://github.com/zephyrproject-rtos/zephyr/pull/42063 should unblock things.  I haven't done a full twister run yet though.

PR #42063 (2022-01-24): soc/intel_adsp: Fixups for cavs18 SMP boot
This platform was stale for a long time and got a little left behind.
Basic OS stuff was working but secondary core bringup didn't.  It has
a slightly different set of choices from the "weird hardware
quirks" menu:

+ Like cAVS 1.5, it boots from a ROM that needs a short delay after
  power-up before it can receive the startup IDC.

+ But, like 2.5 and unlike 1.5, it doesn't start running until the
  PWRCTL bit for the core gets set by DSP software (1.5 gets launched
  by the host).  So the delay needed to move down a bit.

+ Uniquely on this platform [... cut]
closed 2022-01-24 by nashif