zephyrproject-rtos/zephyr #42011: Establish guidelines for TSC working groups

nashif, 2022-01-20:
Establish and document guidelines for each TSC Working group:
- Define Criteria for creating a new group
- Define process for electing/assigning a chairperson
- Define charter for each TSC WG
- How discussions and recommendations are documented and where they are archived
- When does a WG need to report back to the TSC about progress
- Escalation path to the TSC and when discussions are moved to the TSC (in case of a deadlock)
- Identify different types of WGs:
  - Short term vs Long term running WGs
  - WG that work with other committees (Security, Safety, etc.)
- Define general and common process for all WGs

nashif, 2022-01-20:
@mbolivar-nordic  FYI

mbolivar-nordic, 2022-01-21:
We can take this in the process WG, but I guess the outcome is something that we will need to take to the TSC for approval, right? Otherwise we'd be putting the cart before the horse a bit, since process WG is itself a TSC working group.

mbolivar-nordic, 2022-03-16:
Process WG minutes:

A working group (WG) is ... 

- a permanent collaboration serving specific project goals
- established by the TSC
- authorized by the TSC to work on specific area(s)
- dissolved by the TSC when the TSC decides they have become obsolete
- chaired by a project contributor appointed by the TSC

A working group has ...

- a regularly scheduled meeting
- minutes posted for each meeting
- attendees posted for each meeting
- an agenda for the meeting posted 12 hours in advance
- a GitHub project tracking current, proposed, and concluded activities
- a charter document specific to the working group describing its structure and goals, and maintained by the TSC
- a landing page pointing to the charter document on the zephyr wiki
- the authority to make recommendations to the TSC on specific issues within its charter on an as-needed basis as decided by the chair

The Zephyr Project TSC will ...

- consolidate minutes for all working groups in one place
- consider recommendations made by working group chairs
- receive a report from the WG chair on ongoing activities on at least a quarterly basis
- TBD: consolidate video recordings of working group meetings in one place (AI @nashif to check with Brett)

A working group chair will ..

- include  both predominant and dissenting opinions on specific issues in their reports to the TSC

The TSC may ...

- appoint a new WG chair at any time
- introduce requirements on how to handle specific issues within a WG

closed 2022-05-18 by mbolivar-nordic