zephyrproject-rtos/zephyr #42030: can: "bosch,m-can-base": Warning "missing or empty reg/ranges property"

erwango, 2022-01-21:
**Describe the bug**
When building (any application) one of the boards that include can node with compatible "bosch,m-can-base",
the following warning is generated:
```
~/zephyrproject/zephyr/samples/synchronization/build/nucleo_h723zg/zephyr/zephyr.dts:2708.7-2734.5: Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property
```

**To Reproduce**
`west build -b nucleo_h723zg samples/hello_world`

**Expected behavior**
No warning generated

**Impact**
Annoyance

henrikbrixandersen, 2022-01-21:
Agreed. We need to look into reworking the current hierarchy of Bosch M_CAN devicetree bindings (e.g. for STM32, we have `compatible = "bosch,m-can-base"` nodes with `compatible = "st,stm32-fdcan"` children, which is kind of weird).

henrikbrixandersen, 2022-04-21:
@erwango First step in solving this: #45014

henrikbrixandersen, 2022-04-21:
@erwango Second step in solving this: #44951

More to follow.

str4t0m, 2022-08-12:
@henrikbrixandersen have you planned any further changes here?

I have drafted a PR to move STM32FD clock configuration from Kconfig to dts.
It would add the config to the instance node(can1, can2), although this is a property shared between can instances.
But the parent node is generic and would need bigger refactoring to be able to accommodate the very specific STM32 clock setup. [WIP](https://github.com/str4t0m/zephyr/tree/can_stm32fd_clock_control)

Therefore it would be interesting to hear about any progress or plans from your side, before I open a PR or rework this.

henrikbrixandersen, 2022-08-12:
> @henrikbrixandersen have you planned any further changes here?

Yes! :)

> I have drafted a PR to move STM32FD clock configuration from Kconfig to dts. It would add the config to the instance node(can1, can2), although this is a property shared between can instances. But [...]

Putting the clocks property within each child node goes hand-in-hand with my planned changes. I want us to get rid of the parent node altogether and move towards the Linux kernel devicetree bindings for the Bosch M_CAN IP.

> Therefore it would be interesting to hear about any progress or plans from your side, before I open a PR or rework this.

I started the work on transitioning the bindings a while back, but never got around to finishing it. I plan to pick it up again for it to be included in Zephyr v3.2. The major change is that the current way of determining the memory layout needs to be changed from being static and shared among all instances to being a per-instance base address and offsets in order to work with the individual dts nodes/new bindings.

str4t0m, 2022-09-04:
> Putting the clocks property within each child node goes hand-in-hand with my planned changes. I want us to get rid of the parent node altogether and move towards the Linux kernel devicetree bindings for the Bosch M_CAN IP.

Great, I have now opened #49884 to move clock configuration from Kconfig to dts.

mbolivar-nordic, 2022-09-15:
> I plan to pick it up again for it to be included in Zephyr v3.2.

@henrikbrixandersen is this still the plan?

PR #58319 (2023-05-29): drivers: can: mcan: use per-instance message RAM configuration
Switch the Bosch M_CAN devicetree binding to use a bosch,mram-cfg property for specifying the memory layout of the Bosch M_CAN Message RAM. This is  identical to the Linux kernel devicetree binding for Bosch M_CAN IP core based CAN controllers. This introduces an offset cell which can be used for controllers with shared Message RAM between Bosch M_CAN instances.

Restructure the Bosch M_CAN driver backend to use per-instance Message RAM configuration. This removes the need for a common, artificial "can" devicetree node for SoCs wit [... cut]
commit 0f36f1a3 (2023-05-29): drivers: can: mcan: use per-instance message RAM configuration

Restructure the Bosch M_CAN driver backend to use per-instance Message RAM
configuration.

This removes the need for a common, artificial "can" devicetree node for
SoCs with multiple Bosch M_CAN-based CAN controllers and allows for
per-instance configuration of the number of e.g. standard (11-bit) and
extended (29-bit) filter elements.

As part of the restructure, software handling of CAN filter flags was moved
from per-flags bitfields to per-filter bitfields, solving an issue when
using more than 32 standard (11-bit) filter eleme [... cut]
closed 2023-05-29 by nashif