zephyrproject-rtos/zephyr #42048: Add subtree name for settings

Thalley, 2022-01-21:
**Is your enhancement proposal related to a problem? Please describe.**
The API to store something in settings is something like (from https://docs.zephyrproject.org/latest/reference/settings/index.html#example-persist-runtime-state)
```
settings_save_one("foo/bar", &foo_val, sizeof(foo_val));
```
but to extract it a handler is required like
```
struct settings_handler my_conf = {
    .name = "foo",
    .h_set = foo_settings_set
};
``` 
which has a `.name` field for the "subtree", and then to compare the setting name value something like 
```
if (settings_name_steq(name, "bar", &next) && !next) {
``` 
is needed, which basically takes the `"foo/bar"` and split this into two values when extracting/loading the data. 
This is subpar as the string you use to store data is not the same as you use to retrieve data. 
Using and comparing string literals (like `"foo/bar"`, `"foo"`, `"bar"`) is a poor design choice compared to using variables or even macros, as that can much easier avoid any typo issues and reduce issues during refactors. 

The current API does not support using macros or variables very well, precisely because it does not use the same save value to store as to retrieve

**Describe the solution you'd like**
The way values are stores and retrieved should be the same. The simplest solution is to add the "subtree" value to the `settings_save_one` (and other APIs that take a single string), so that 
```
settings_save_one("foo/bar", &foo_val, sizeof(foo_val));
```
becomes something like 
```
int settings_save_one(const char *subtree_name, const char *key_name, const void *value, size_t value_len);
```

so that we can do

```
const char foo_str = "foo"; // or #define FOO_STR "foo"
const char bar_str = "bar"; // or #define BAR_STR "bar"
...
settings_save_one(foo_str, bar_str, &foo_val, sizeof(foo_val));
...
if (settings_name_steq(name, bar_str, &next) && !next) {
```

**Describe alternatives you've considered**
The alternative is of course to do the opposite and instead of something like 
```
    if (settings_name_steq(name, "bar", &next) && !next) {
```
we do
```
    if (settings_name_steq(name, "foo/bar", &next) && !next) {
```
so that we can do

```
const char foo_bar_str = "foo/bar"; // or #define FOO_BAR_STR "foo/bar"
...
settings_save_one(foo_bar_str, &foo_val, sizeof(foo_val));
...
if (settings_name_steq(name, foo_bar_str &next) && !next) {
```

**Additional context**
N/A

Laczen, 2022-01-21:
@Thalley, the reason settings doesn't work like this is that the set handler can operate on an entire different level compared to the save. This makes that you can have one handler for all `bt` related data, and this one handler can then call different set routines based on the rest of the name.

I don't quite get why the present solution does not support macro's very well. It is just needed to insert the separator.

Thalley, 2022-01-24:
> @Thalley, the reason settings doesn't work like this is that the set handler can operate on an entire different level compared to the save. This makes that you can have one handler for all `bt` [...]

It is possibly by using two macros (which is also the solution we came up with in https://github.com/zephyrproject-rtos/zephyr/pull/40626), but nonetheless the enhancement still stands. Using two different values for store and retrieve is uncommon for a key-value database like this - Or rather splitting the key for retrieve is.

Laczen, 2022-01-24:
@Thalley, the key isn't split for retrieve, the routines are just advancing every step down. The disadvantage of using the `settings_save_one(subtree, key, ...)` is that the final key needs to be created on stack, while now there are backends like nvs that do nothing with this key but to compare and if required store. No creation of a final key is required.

While having a quick look at the PR I am even wondering why a special handler is created for the `appearance`, it could just be a part of the `bt` handler.

Thalley, 2022-01-24:
> The disadvantage of using the settings_save_one(subtree, key, ...) is that the final key needs to be created on stack, while now there are backends like nvs that do nothing with this key but to compare and if required store. No creation of a final key is required.

That is a good point, and certainly a disadvantage of this approach, unless the final key doesn't need to be combined before we actually write to flash or where, then no additional stack variables to hold that is necessary. 

> While having a quick look at the PR I am even wondering why a special handler is created for the appearance, it could just be a part of the bt handler.

I think that may have been too quick of a look; the `appearance` is using the generic `bt` handler :)

Laczen, 2022-01-24:
> I think that may have been too quick of a look; the `appearance` is using the generic `bt` handler :)

Yes it was to quick a look :).

To simplify the use of the macro's it would be possible to create 3 macro's:
a. For the subtree: `#define SETTINGS_SUBTREE "bt"`,
b. For the key: `#define SETTINGS_KEY_APPEARANCE "appearance"`,
c. For the key name: `#define SETTINGS_KEY_APPEARANCE_NAME SETTINGS_SUBTREE"\"SETTINGS_KEY_APPERANCE",

BTW during the rework of settings some years ago I have asked to add the separator to the handler name,
this might have made the creation of key names easier. It would also have been simpler in the processing of the name during retrieve since the separator would have been part of the compare.

Thalley, 2022-01-24:
> Yes it was to quick a look :).
>
> To simplify the use of the macro's it would be possible to create 3 macro's: a. For the subtree: `#define SETTINGS_SUBTREE "bt"`, b. For the key: `#define [...]

Yes, that is a workaround, but it still not I have requested in this issue, which is the ability to use the same key(s) for store and retrieve, i.e. 
```
const char foo_str = "foo"; // or #define FOO_STR "foo"
const char bar_str = "bar"; // or #define BAR_STR "bar"
...
settings_save_one(foo_str, bar_str, &foo_val, sizeof(foo_val));
...
if (settings_name_steq(foo_str, bar_str, &next) && !next) {
```

Which is the most common way of dealing with a key-value database, as this is.

Laczen, 2022-01-24:
> Yes, that is a workaround, but it still not I have requested in this issue, which is the ability to use the same key(s) for store and retrieve, i.e.
> 
> ```c
> const char foo_str = "foo"; // or #define FOO_STR "foo"
> const char bar_str = "bar"; // or #define BAR_STR "bar"
> ...
> settings_save_one(foo_str, bar_str, &foo_val, sizeof(foo_val));
> ...
> if (settings_name_steq(foo_str, bar_str, &next) && !next) {
> ```
> 
> Which is the most common way of dealing with a key-value database, as this is.

Because the retrieve is not done by searching the key. It can be done by using `settings_load_subtree()`, but it is not in general. If you step down one level from the handler the subtree would need to be inherited from the above handler. Since you would need to construct the subtree anyhow it is a small step to introduce the macro's as you proposed.

While a method of handling settings storage as you proposed has been requested in the past nobody has volunteered to write such a system. It would probably make support for run time generated keys a lot more difficult.

Thalley, 2022-01-24:
> Because the retrieve is not done by searching the key. 

And that's the part that seems somewhat odd to me; storing by (aggregated) key, but retrieving with (split) key. 

> While a method of handling settings storage as you proposed has been requested in the past nobody has volunteered to write such a system.

That sounds like reasoning enough to keep this enhancement open. Are there other enhancements for this open somewhere?

> It would probably make support for run time generated keys a lot more difficult.

I don't see how. Can you elaborate?

Laczen, 2022-01-24:
@Thalley, the "enhancement" you propose is very limited in functionality as it only makes sense at the handler level and with the disadvantage of requiring the storage key to be created on stack. The requested functionality can easily be achieved using macros.

In the settings system the storing is completely separated from the retrieving and this makes it very versatile (albeit a little difficult to understand at first).

If you would like you could use the settings system exactly as you propose, you register a handler for each variable you need e.g. `\bt\myvars\myvar1` and you save the data under this variable. Not all users should be obliged to use it this way.

Thalley, 2022-01-25:
> with the disadvantage of requiring the storage key to be created on stack.

Only as it is implemented now. The storage key does not need to be assembled at all before actually writing to the storage, does not?

> The requested functionality can easily be achieved using macros.

I would call that a workaround, as yes we can use macros to simulate what we want, i.e. reusing the same macros for storing and retrieving, but we won't be able to use any variables. 

> f you would like you could use the settings system exactly as you propose, you register a handler for each variable you need e.g. \bt\myvars\myvar1 and you save the data under this variable. Not all users should be obliged to use it this way.

That is a possibility, yes. 

What is equally interesting is that the `settings_runtime_set` and `settings_runtime_get` actually does exactly what I have been requesting (although with a aggregated key, instead of a split key). Do you know what sideeffects or disadvantages (if any) the `_runtime_` functions have, compared to the others? They seem significantly more intuitive to use and much more flexible, at least for individual variables. I assume there's a performance gain on using the handlers for subtrees, if retrieving multiple variables.

Laczen, 2022-01-25:
> > with the disadvantage of requiring the storage key to be created on stack.
> 
> Only as it is implemented now. The storage key does not need to be assembled at all before actually writing to the storage, does not?

Yes it does need to be assembled if it is given in parts, it might be possible to write it incrementally but it might not be. By definition a preassembled key does not need to be assembled. Some backends might assemble the entry as being the key + the value, some backends separate the key from the value to avoid rewriting the keys.
 
> What is equally interesting is that the `settings_runtime_set` and `settings_runtime_get` actually does exactly what I have been requesting (although with a aggregated key, instead of a split key). [...]

If you find it more convenient to use the `_runtime_` functions use them. The settings system has a lot of usage options, select what fits you best. The code is there to see if it will be efficient or not.

Thalley, 2022-01-25:
@alwa-nordic Do you have any inputs? Would the `settings_runtime_` handle this enhancement request? 

Personally I think we can probably close this, given the existing flexibility of the settings API. Thanks for your input @Laczen

alwa-nordic, 2022-01-26:
> Would the settings_runtime_ handle this enhancement request?

I believe we use the `settings_load` API because it is optimized better. `settings_runtime_` does solve this problem. I don't know if it is ok to use it in the stack.

The most non-intuitive bit of API in my eyes is the asymmetry between `h_export(export_func(name, ...))` and `h_set(key, ...)`. In this API `name` is the full path while `key` is a relative path.

https://docs.zephyrproject.org/latest/reference/settings/index.html?#c.settings_handler_static

Laczen mentioned that this it's like this to avoid allocating strings in the settings subsystem. But I'm not sure allocating is necessary. It would just be two writes to a buffer instead of one.

My ideal API change is that `h_export` gets the `struct settings_handler` or an equivalent as a parameter and that `export_func` / `settings_save_one` takes this parameter and uses relative paths.

This way, it's the settings handlers become truly path-independent. E.g. we could have `/lightbulbs/0/brightness` and `/lightbulbs/1/brightness` and have `/lightbulbs/0` and `/lightbulbs/1` be handled by the same definition of `h_set` and `h_export`.

But I'm not sure this is worth breaking the public API for.

Laczen, 2022-01-26:
> The most non-intuitive bit of API in my eyes is the asymmetry between `h_export(export_func(name, ...))` and `h_set(key, ...)`. In this API `name` is the full path while `key` is a relative path.

The settings system is not a key-value storage system. What the settings system does is declare how to process a key into some setting of the system. The settings system has no knowledge about what is stored in the storage backend, it only knows how to process things.

It is not about relative path versus full path, the settings_handler name is just the start of the key that is processed. As this start is known as it is calling the handler there is no longer any use of this start.

The h_export() functionality is not using any data in the backend at all, so it can only go over all registered handlers using their name.

The representation of relative vs absolute path is an interpretation, it has nothing to do with that. 
 
> Laczen mentioned that it's like this to avoid allocating strings in the settings subsystem. But I'm not sure allocating is necessary. It would just be two writes to a buffer instead of one.

?? And how exactly are you going to do 2 writes to a buffer without allocating a buffer ?

Laczen, 2022-01-26:
> > The most non-intuitive bit of API in my eyes is the asymmetry between `h_export(export_func(name, ...))` and `h_set(key, ...)`. In this API `name` is the full path while `key` is a relative path.
> 
> The settings system is not a key-value storage system. What the settings system does is declare how to process a key into some setting of the system. The settings system has no knowledge about what is stored in the storage backend, it only knows how to process things.
> 
> It is not about relative path versus full path, the settings_handler name is just the start of the key that is processed. As this start is known as it is calling the handler there is no longer any use of this start, so the key is advanced after the start.
> 
> The h_export() functionality is not using any data in the backend at all, so it can only go over all registered handlers using their name.
> 
> The representation of relative vs absolute path is an interpretation, it has nothing to do with that.
> 
> > Laczen mentioned that it's like this to avoid allocating strings in the settings subsystem. But I'm not sure allocating is necessary. It would just be two writes to a buffer instead of one.
> 
> ?? And how exactly are you going to do 2 writes to a buffer without allocating a buffer ?

alwa-nordic, 2022-01-28:
> The settings system is not a key-value storage system. 

From https://docs.zephyrproject.org/latest/reference/settings/index.html:

> Settings items are stored as key-value pair strings. By convention, the keys can be organized by the package and subtree defining the key, for example the key id/serial would define the serial configuration element for the package id.

Sounds very much like it's sold as a key-value storage system.

> > Laczen mentioned that it's like this to avoid allocating strings in the settings subsystem. But I'm not sure allocating is necessary. It would just be two writes to a buffer instead of one.
>
> ?? And how exactly are you going to do 2 writes to a buffer without allocating a buffer ?

I assume all the backends are serializing the key and the value into one stream. The key can be in two parts. I do not appreciate your '??'. It's rude.

Laczen, 2022-01-28:
> @Laczen
> 
> > The settings system is not a key-value storage system.
> 
> From https://docs.zephyrproject.org/latest/reference/settings/index.html:
> 
> > Settings items are stored as key-value pair strings. By convention, the keys can be organized by the package and subtree defining the key, for example the key id/serial would define the serial configuration element for the package id.
> 
> Sounds very much like it's sold as a key-value storage system.

In a key-value storage system the key is not only used to store, but also to retrieve the value.

In the settings system the key is not used to retrieve the value, instead for all stored key-value combination a method is defined to use the value. The settings system does not load a key-value combination.
 
> I assume all the backends are serializing the key and the value into one stream. The key can be in two parts. I do not appreciate your '??'. It's rude.

I'm sorry if you considered this as rude. But as I explained above your assumption is wrong.

Thalley, 2022-01-28:
> In a key-value storage system the key is not only used to store, but also to retrieve the value.
>
> In the settings system the key is not used to retrieve the value, instead for all stored key-value [...]

Is that exactly what the `settings_runtime_get` function does?

Laczen, 2022-01-28:
> Is that exactly what the `settings_runtime_get` function does?

Not by default, `settings_runtime_get` just calls the `h_get()` routine that is associated with a handler.  This could allow you to retrieve present values of some key-value combination. This does not do anything on the backend, so it doesn't retrieve any stored information.

Thalley, 2022-01-28:
> Not by default, `settings_runtime_get` just calls the `h_get()` routine that is associated with a handler. This could allow you to retrieve present values of some key-value combination. This does [...]

This issue was raised for the settings API, and not so much the backend. The API certainly exposes settings as key-value storage (and I am perfectly fined with closing this, as that is what I was really suggesting to add, before I found the `settings_runtime_` API). What is going on in the backend is irrelevant to my original request :)

Laczen, 2022-01-28:
@Thalley, just to be extra clear and avoid any confusion, the `settings_runtime_` API does not retrieve anything from storage.

Thalley, 2022-01-28:
> @Thalley, just to be extra clear and avoid any confusion, the `settings_runtime_` API does not retrieve anything from storage.

Then what does it do? 

The brief is `Get a value corresponding to a key from a module handler.` and the `example_runtime_usage` shows an example of injecting (odd word of choice) into settings, and then also fetching the value again. 

The implementation calls `h_get` which has this brief: `Get values handler of settings items identified by keyword names.`, so I am a bit confused. 

Can you elaborate on what it does then?

Laczen, 2022-01-28:
> Then what does it do?
>
> The brief is `Get a value corresponding to a key from a module handler.` and the `example_runtime_usage` shows an example of injecting (odd word of choice) into settings, and [...]

Suppose you have a key: `bt/id` and this id is changed during use, and we call the variable `id`
During startup you would like to restore a value: you define a handler with name `bt/id` with handlers `h_set` and `h_get`. `h_set` is called when during `settings_load()` stored item with key=`bt/id` is found.
Now during runtime there is some other part of the application that needs to now the value of id, the routine `settings_runtime_get("bt/id", ...)` is used, in that case the `h_get` routine will be called to return the present "runtime" value of id. It would be needed to specify explicitely in the `h_get` routine how the value is returned.

Hope this clears up confusion.

Thalley, 2022-01-28:
> Hope this clears up confusion.

Sure does, thanks! That isn't particularly clear from the documentation, nor the example though. 

However in that case, then the original request from this issue still stands :)

Laczen, 2022-01-28:
> However in that case, then the original request from this issue still stands :)

Yes it does :), but for settings the `subtree` has no meaning whatsoever, the meaning is defined in the bt subsystem. So it would be best that the bt subsystem assembles the key (either  at compile time using macro's, or in runtime using a buffer).

Thalley, 2022-01-28:
> Yes it does :), but for settings the `subtree` has no meaning whatsoever, the meaning is defined in the bt subsystem. So it would be best that the bt subsystem assembles the key (either at compile [...]

We did talk about that solution as well, and basically implement more `bt_settings_` APIs. 

> but for settings the subtree has no meaning whatsoever,

Doesn't all settings need a `foo/bar`, i.e. a `/`, or can you just store values as `foo` and `bar`?

Laczen, 2022-01-29:
> > but for settings the subtree has no meaning whatsoever,
> 
> Doesn't all settings need a `foo/bar`, i.e. a `/`, or can you just store values as `foo` and `bar`?

You can store values under `foo` and `bar` without a `/`.

Thalley, 2022-01-31:
> > > but for settings the subtree has no meaning whatsoever,
> > 
> > 
> > Doesn't all settings need a `foo/bar`, i.e. a `/`, or can you just store values as `foo` and `bar`?
> 
> You can store values under `foo` and `bar` without a `/`.

Okay, then there has been a fundamental misunderstanding on our side :) This is probably based on the significant use of the word `subtree` in `settings.h`, but if there is no such thing as a `subtree` in the settings, and that they are all just "keys", then the documentation need a significant update.

Laczen, 2022-01-31:
> Okay, then there has been a fundamental misunderstanding on our side :) This is probably based on the significant use of the word `subtree` in `settings.h`, but if there is no such thing as a [...]

That is OK :).

Regarding the use of `subtree` in settings.h you are misunderstanding a `helper` feature with a requirement. The settings system contains very few requirements, but quite a lot of helpers e.g. subtree loading (which is just ignoring keys that do not start with the subtree), `settings_name_steq`, `settings_name_next`. Al these helpers work in cooperation with the 'separator /'. There is even a helper that allows you to load keys that don't have a handler (`settings_load_subtree_direct`).

Not all functionality is equally documented. What is surely missing is the documentation on how the `bt` subsystem is using the settings system (but this shouldn't be part of the settings documentation).

Thalley, 2022-01-31:
> That is OK :).
>
> Regarding the use of `subtree` in settings.h you are misunderstanding a `helper` feature with a requirement. The settings system contains very few requirements, but quite a lot of [...]

I think you may have missed my point. You stated 
> but for settings the subtree has no meaning whatsoever

But that seemingly isn't the case, with the significant use of the term in the settings API. 

You also stated 
> You can store values under foo and bar without a /.

So if I store a value 

```
settings_save_one("foo", &foo_val, sizeof(foo_val));
```

Then there's only the key "foo" and no subtree. 

However, to get the value from storage, I'd need to use a `settings_handler`, right? Which would be something like 
```
struct settings_handler my_conf = {
    .name = "foo",
    .h_set = foo_settings_set
};
```

and then also have 
```
static int foo_settings_set(const char *name, size_t len,
                            settings_read_cb read_cb, void *cb_arg)
{
    int rc;
    
    rc = read_cb(cb_arg, &foo_val, sizeof(foo_val));
    if (rc >= 0) {
        /* do stuff with foo_val */
        return 0;
    }

    return rc;
}
```

In this example there is no subtree, just a key "foo", and according to you've said, that should work right? 
So the documentation mentions "key" and "subtree", and we can store a value using a key "foo" that doesn't have a subtree. 

My, and I guess also @alwa-nordic, point is that it would be a much simpler API, and easier to understand, if the API simply used a `settings_save_one(key, value, value_len)`, and a `settings_retrieve_one(key, value, value_len)`.

Laczen, 2022-01-31:
> My, and I guess also @alwa-nordic, point is that it would be a much simpler API, and easier to understand, if the API simply used a `settings_save_one(key, value, value_len)`, and a `settings_retrieve_one(key, value, value_len)`.

Probably this would be simpler to understand, but also less efficient on a slow backend. And also a lot less flexible. If you would like such the routine `settings_retrieve_one(key, value, value_len)` to be added I see no problem with creating one, but it would need the side note that this requires each retrieve to go through the entire set of stored keys and could result in slow behavior. Of course this should also require a well defined error return value ("invalid key", "invalid length", ...) and correct handling of this error by the user.

BTW, this doesn't change anything regarding the use of `subtree`.

Thalley, 2022-01-31:
> Probably this would be simpler to understand, but also less efficient on a slow backend. And also a lot less flexible. If you would like such the routine `settings_retrieve_one(key, value, [...]

But can't the same be said for the `settings_save_one`? Isn't it also slower to write only a single value, instead of a subtree? 

> BTW, this doesn't change anything regarding the use of `subtree`.

That depends on whether you still claim that `subtree` isn't part of the settings, or if it is. If all settings need a `subtree`, then the save/load (or retrieve) should contain a subtree argument. But if `subtree` isn't required, then there is no subtree, just a `key` which just happens to have a `/` in the name. 

Btw, I appreciate your fast responses to this matter :) I would, however, like to get input from another person as well. Are there others that may have some input here?

Laczen, 2022-01-31:
> But can't the same be said for the `settings_save_one`? Isn't it also slower to write only a single value, instead of a subtree?

Writing a value requires an item to be added at the end, not going through the backend to find the value. When writing multiple values there is already the possibility to use the export function that can avoid opening/closing of the backend (although no backend is using this at the moment).

> That depends on whether you still claim that `subtree` isn't part of the settings, or if it is. If all settings need a `subtree`, then the save/load (or retrieve) should contain a subtree argument. [...]

It seems to me like you would like the subtree to be a requirement for the settings subsystem. What would be:
a. The definition of the subtree ? Should it be limited to the handler (so each subtree would have it's own handler), or should it be something else ?
b. Should it be allowed to define handlers at different levels (e.g. subtree `bt`, subtree `bt/btmesh`, ...) and how should they be handled ?
c. Does this have any advantage compared to not having the subtree at all ?

Thalley, 2022-01-31:
> Writing a value requires an item to be added at the end, not going through the backend to find the value. When writing multiple values there is already the possibility to use the export function that can avoid opening/closing of the backend (although no backend is using this at the moment).

So wait, if I call 
```
settings_save_one("foo", &foo_val, sizeof(foo_val));
```
twice, then there will be 2 values stored, i.e. it **appends** data instead of overwriting existing value for the same key? 

> It seems to me like you would like the subtree to be a requirement for the settings subsystem. 

I want there to be either a subtree and a key pair (e.g. ("bt", "name"), or just a key "bt/name", but the current API uses both which is confusing. Personally, for the sake of simplicity and how most other key/value storages work, I would go for just a key, although that would limit reads and writes to a single value instead of multiple at once (which may hurt performance). 

> a. The definition of the subtree ? Should it be limited to the handler (so each subtree would have it's own handler), or should it be something else ?

I don't have a proper definition of subtree, and doing a fast attempt at defining it may not be a great idea, but it is nonetheless used in a specific way in the `settings.h` file. 

> b. Should it be allowed to define handlers at different levels (e.g. subtree bt, subtree bt/btmesh, ...) and how should they be handled ?

If we go subtree idea, then we could have recursive subtrees. Not sure what you mean by "how should they be handled", as we are already handling that in the current implementation. 

c. Does this have any advantage compared to not having the subtree at all ?

If bulk loading/storing can be done, then there is a performance gain, otherwise not really. 

d. What are the things that could be done that are not possible now ?

Nothing. The current API seemingly supports all the cases. 

In reality, loading a value from settings (which is a key-value storage) isn't (or shouldn't) be much different than loading a file in a filesystem. A file basically contains a `parent` and a `name` and the path to it is thus usually something like `parent/name` or `parent\name`, which is also similar to many REST APIs as well query languages such as XPath or XQuery.

Laczen, 2022-01-31:
> So wait, if I call
>
> ```
> settings_save_one("foo", &foo_val, sizeof(foo_val));
> ```
>
> twice, then there will be 2 values stored, i.e. it **appends** data instead of overwriting existing value for the [...]

No, the backend will normally filter this out to avoid flash wear. But this is not really important, if the backend filters out duplicate writes it needs to do this for each value that is saved, regardless of a bulk store or not.

> I want there to be either a subtree and a key pair (e.g. ("bt", "name"), or just a key "bt/name", but the current API uses both which is confusing. Personally, for the sake of simplicity and how [...]

This is how the `bt` subsystem is using the API, it is not enforced to do so. But it is efficient doing it the way it is done. Requiring very few error handling and very small intermediate buffers as it first verifies the data size before reading it.

> I don't have a proper definition of subtree, and doing a fast attempt at defining it may not be a great idea, but it is nonetheless used in a specific way in the `settings.h` file.
>
>
> If we go [...]

This functionality is provided, but on top of this there is a method to "process" a directory and to step down from this directory level.

Regarding the `subtree` functionality in settings, this used not to be there, but it has great value in places where you only wish to update some variables: suppose you have some bt connection values that are only for a specific client and you wish to set them when a client connects. The settings are stored under `bt\client_id`. Now when this client connects (you know it's id) you call `settings_load_subtree("bt\client_id")`, this will go trough the storage but only when it matches `bt\client_id` the system calls the (bt) set method.

Laczen, 2022-01-31:
OK, I'm out. I will comment on whatever PR that comes in.

Thalley, 2022-01-31:
> OK, I'm out. I will comment on whatever PR that comes in.

Thanks for your time so far

Laczen, 2022-02-01:
@nvlsianpu, @de-nordic.