zephyrproject-rtos/zephyr #42050: printk bug: A function called from printk is invoked three times given certain configuration variables

jeback1, 2022-01-21:
**Describe the bug**
Calling a function from a printk statement invokes the function three times.

This depends on having both of these two configuration variables set true: 
```
CONFIG_LOG=y
CONFIG_LOG_PRINTK=y
```

**To Reproduce**
Steps to reproduce the behavior:
prj.conf:
```
CONFIG_NEWLIB_LIBC=y
CONFIG_LOG=y
CONFIG_LOG_PRINTK=y
```
main.c:
```#include <zephyr.h>
#include <stdio.h>
#include <stdlib.h>

static int my_int = 0;
int my_func(void){
	my_int++;
	printf("\t\t in my_func; my_int=%d\n",my_int);
	return my_int;
}

void main(void)
{
	printf("\n testing printf\n");
	printf("\t printf: value returned by my_func: %d\n", my_func());
	printf("\n testing printk\n");
	printk("\t printk: value returned by my_func: %d\n", my_func());
}
```
build:
```
source ~/zephyr/zephyr/zephyr-env.sh
west build -b native_posix
west build -t run
```
output:
```
*** Booting Zephyr OS build v2.7.99-3561-g25e6803705b2  ***

 testing printf
		 in my_func; my_int=1
	 printf: value returned by my_func: 1

 testing printk
		 in my_func; my_int=2
		 in my_func; my_int=3
		 in my_func; my_int=4
	 printk: value returned by my_func: 4

```
**Expected behavior**
I expect the function to be invoked only once. This is the case with either CONFIG_LOG=n or CONFIG_LOG_PRINTK=n:

prj.conf:
```
CONFIG_NEWLIB_LIBC=y
CONFIG_LOG=n
CONFIG_LOG_PRINTK=y
```
output:
```
*** Booting Zephyr OS build v2.7.99-3561-g25e6803705b2  ***

 testing printf
		 in my_func; my_int=1
	 printf: value returned by my_func: 1

 testing printk
		 in my_func; my_int=2
	 printk: value returned by my_func: 2
```

**Impact**
It confused the hell out of me. I only discovered it because I had a left-over print statement in a function being called from LOG_INF. 

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: ~/gnuarmemb/gcc-arm-none-eabi-9-2019-q4-major
 - Version: Zephyr OS build v2.7.99-3561-g25e6803705b2

**Additional context**
My target microcontroller is the nrf52840. I get the same unexpected behavior when building and running on nrf52840dk.

This unexpected behavior also occurs with the LOGGING module, which I'm guessing uses printk.

**The test program**
[test_printk.zip](https://github.com/zephyrproject-rtos/zephyr/files/7915256/test_printk.zip)

nordic-krch, 2022-01-26:
@jeback1 you can try following fix #42165.

nordic-krch, 2022-02-10:
reopening since fix got reverted #42609.

PR #42686 (2022-02-22): logging: Prevent multiple arguments evaluation (take 2)
This is another attempt to fix #42050. First attempt (#42165) was reverted (#42609) due to #42608. Test was failing because of increased stack usage after changes in logging. Increased stack usage has been seen in `CONFIG_NO_OPTIMIZATIONS=y` case.

In this take, logging macro path has been changed so that runtime packaging (the one without preprocessor magic and multiple arguments evaluation) is used always when `CONFIG_NO_OPTIMIZATIONS=y` or `CONFIG_LOG_MODE_IMMEDIATE=y`. In both cases static packaging is not needed since static packagin [... cut]
closed 2022-02-22 by carlescufi