zephyrproject-rtos/zephyr #42066: Planned USB device/host support rework/enhancements for 2022

jfischer-no, 2022-01-22:
## Introduction

Currently we have only USB device support in Zephyr OS. Existing USB device support has many disadvantages and issues, like missing support for multiple driver instances, no endpoint (request) buffers management, missing class instances management and possibility to configure/enable class instance at runtime, not a good notification channel to the user/application, messy callback architecture, no API to set/reconfigure idVendor/iSerialNumber/bcdDevice at runtime, confusing usb_dc_ep_read()/usb_dc_ep_write() API, and more.

There is no USB host support in the project. It is asked again and again for it and there was effort like #30361, but that is far from suitable. USB device stack would also benefit from the host support and allow self-contained tests with emulated device and host controllers, and better support for USBIP.

Since many problems with the USB device support are due to the way it is designed (imported in Zephyr OS), we would like to redesign USB device support. That means USB device controller driver API and USB device stack will be rewritten in parallel and independent to existing implementation. Due to the need for testing/validation, USB host controller driver API and some parts of higher level will also be implemented.

## Rough plan

- [ ] Introduction of USB device controller API and a device controller driver implementation (#38679)
- [ ] New USB device stack implementation (WIP)
- [ ] Porting more device controller drivers to new API (-)
- [ ] Introduction of USB host controller API and a host controller driver implementation (WIP)
- [ ] Porting USBIP driver to new APIs (-)
- [ ] Porting USB device classes to new USB device support (-)

koffes, 2022-01-24:
This is good news. I would like to link to this issue: https://github.com/zephyrproject-rtos/zephyr/issues/26857, and hopefully the high CPU load will be resolved here as well.
Also, any plans for async USB audio support?

qipengzha, 2022-01-25:
For driver support, will you plan to enable Synopsys DWC USB2.0 OTG controller ? thx

dleach02, 2022-03-02:
Do we need a separate tracking ticket for USB Host?

dleach02, 2022-03-09:
@jfischer-no Is there any of this that can be parallelized if additional resources are made available?

jfischer-no, 2022-03-10:
> @jfischer-no Is there any of this that can be parallelized if additional resources are made available?

Not yet.

jfischer-no, 2022-04-29:
Plan updated.

lenghonglin, 2022-07-13:
hello, i use usb audio these , and the usb audio leak some feature unit, like volume . It there have some plan to do this?

troica65, 2022-08-11:
Hi, any estimate for when host support will be available ?

jfischer-no, 2022-08-15:
>hello, i use usb audio these , and the usb audio leak some feature unit, like volume . It there have some plan to do this?

Yes, but it is not directly relevant to this topic.

> Hi, any estimate for when host support will be available ?

ASAP!

jfischer-no, 2022-10-05:
Virtual controller and host support has been separated from #38679  and moved to #50985. Updated the list of drivers and class implementations to be ported.

raveslave, 2022-12-06:
@MaureenHelm any interest from NXP to support the mcux usb host (and device) drivers for LPC and i.MXRT in zephyr?

koffes, 2022-12-06:
Wrt. Audio. Could someone please enlighten me on the status/plans for async USB support? We need the SoC to poll the PC host for new Audio data.

jfischer-no, 2022-12-06:
> Wrt. Audio. Could someone please enlighten me on the status/plans for async USB support? We need the SoC to poll the PC host for new Audio data.

I commented to a similar question above. It would be nice if you open a well-described enhancement issue or discussion, but it does not belong here!

> @MaureenHelm any interest from NXP to support the mcux usb host (and device) drivers for LPC and i.MXRT in zephyr?

It appeared to me in a dream that progress on NXP USB controller support is inversely proportional to how often you ask the same question in different topics.

MaureenHelm, 2022-12-06:
> @MaureenHelm any interest from NXP to support the mcux usb host (and device) drivers for LPC and i.MXRT in zephyr?

I don't work at NXP anymore. Please contact @dleach02 for NXP-related questions.

virginie0818, 2023-02-10:
> hello, i use usb audio these , and the usb audio leak some feature unit, like volume . It there have some plan to do this?

i'm also curious when will volume feature be included? thx.

jgl-meta, 2023-04-26:
@jfischer-no What is the status of this? Is this on track to be finished before release 3.4?

jfischer-no, 2023-04-27:
> @jfischer-no What is the status of this? Is this on track to be finished before release 3.4?

Obviously only what is explicitly marked for the v3.4.0 https://github.com/zephyrproject-rtos/zephyr/issues/42066#issue-1111431202

tmon-nordic, 2023-04-28:
Porting all classes is marked for 3.4.0, but considering [feature freeze date (2023/05/26)](https://github.com/zephyrproject-rtos/zephyr/wiki/Release-Management) is just in few weeks it might not be possible.

dleach02, 2023-09-06:
> @MaureenHelm any interest from NXP to support the mcux usb host (and device) drivers for LPC and i.MXRT in zephyr?

Yes, there is interest from NXP and USB host support.

@jfischer-no, what is the right forum to discuss this? Is it here in this ticket?

jfischer-no, 2023-09-14:
> Yes, there is interest from NXP and USB host support.
> 
> @jfischer-no, what is the right forum to discuss this? Is it here in this ticket?

Please not here, maybe _discussions_,  but once you have started implementing a UHC driver, please add a note to the description or ask me for an update. However, the priority is and should be to get the new device support to a state where we can start calling the current stack legacy. This means there should be alternatives to all the class implementations we have for the current stack, and a few more drivers.

dleach02, 2023-09-14:
Okay @jfischer-no. My primary interest is USB host mode. Trying to figure out the pathway to getting this up and running on one of our platforms, maybe the RT5xxx.

hswang914, 2023-09-27:
Hi, I saw the phrase "Introduction of USB device controller API." Does it mean that there is a resource available for me to study and understand the new USB device driver, such as a basic usage guide or a comparison with the old one? Thanks!

MarkWangChinese, 2024-03-14:
Hi @jfischer-no I am working on this task.
- [ ] usb_dc_mcux -> udc_mcux

MarkWangChinese, 2024-03-27:
Hi @jfischer-no I (from NXP) want to contribute the USB host stack because NXP has interest for Zephyr USB host. What can I help for developing the Zephyr USB host stack? If you already have the overall thought about the host stack, I can help to do the implementation. In short, I want to contribute USB Host stack and accelerate its implementation. Thanks. Our team is the owner of the NXP SDK USB stack, so we have a lot of experience on USB2.0.

jfischer-no, 2024-03-27:
> Hi @jfischer-no I (from NXP) want to contribute the USB host stack because NXP has interest for Zephyr USB host. What can I help for developing the Zephyr USB host stack? If you already have the [...]

Nice :heart: . However, here is no independent USB host stack, it is part of the USB support, see my comment https://github.com/zephyrproject-rtos/zephyr/issues/42066#issuecomment-1720138887. Thanks for the upcoming contribution to Zephyr's USB support, especially on ASAP porting NXP USB device driver to new UDC API where each NXP controller type has its own driver and not as clumsy as the existing usb_dc_mcux.

dleach02, 2024-04-01:
@jfischer-no, 

> > Yes, there is interest from NXP and USB host support.
> > @jfischer-no, what is the right forum to discuss this? Is it here in this ticket?
> 
> @dleach02 Please not here, maybe _discussions_, but once you have started implementing a UHC driver, please add a note to the description or ask me for an update. However, the priority is and should be to get the new device support to a state where we can start calling the current stack legacy. This means there should be alternatives to all the class implementations we have for the current stack, and a few more drivers.

If I look at the list of items for moving over to the new UDC it looks like mostly porting vendor's USB drivers. I think it makes sense to work in parallel on the host layer instead of serializing this effort. If we are blocked by other vendors conversions we may never get to the host work. 

We have needs for host support and the resources available to assist in this effort. Just need some guidance on how you want this done.

josuah, 2024-05-07:
I got started with [UVC](https://www.usb.org/document-library/video-class-v15-document-set).

For now a continuation of the Video driver interface. #72311 

The recent UAC2 seems like a good reference for how to implement it. #70029

ddavidebor, 2024-05-17:
is the USB-ip supposed to work with native_sim and the device_next stack?

josuah, 2024-05-17:
> is the USB-ip supposed to work with native_sim and the device_next stack?

If it is about help for troubleshooting, feel free to join the [Discord server](https://discord.gg/zephyrproject), the [Mailing lists](https://lists.zephyrproject.org/g/users), the [Discussions section](https://github.com/zephyrproject-rtos/zephyr/discussions), or open a new issue if you encountered what appears to be a bug.

henrikbrixandersen, 2024-06-15:
> is the USB-ip supposed to work with native_sim and the device_next stack?

Please see https://github.com/zephyrproject-rtos/zephyr/pull/74141

tannewt, 2024-06-25:
Why not use [TinyUSB](https://github.com/hathach/tinyusb) instead of making a new stack? It has device and host support already. It has an OS abstraction layer as well. License is MIT.

Both [Espressif](https://github.com/espressif/esp-usb) and [Raspberry Pi](https://github.com/raspberrypi/pico-sdk/tree/master/lib) use it in their SDKs already.

josuah, 2024-07-03:
> Why not use [TinyUSB](https://github.com/hathach/tinyusb) instead of making a new stack? It has device and host support already. It has an OS abstraction layer as well. License is MIT.

Wonderful work from Ha Thach! Very simple callback-based API that is convenient to debug... Just follow the calls.
Many classes implemented as well. UVC contributed by Toshiba even!

I am not from Nordic and do not know the original rationale for a from-scratch implementation for Zephyr, but here are some clues that I am trying to guess:

- allow to have a Zephyr driver API without too much abstraction in-between
- leverage existing Zephyr code snippets (i.e. net_buf, events, work queues, threads...)
- leverage existing Zephyr subsystems (i.e. CDC_NCM is implementation is mostly glue)
- at this stage, the existing amount of effort already put into it

Just my wild guesses!

Related
- https://github.com/hathach/tinyusb/issues/458

josuah, 2024-08-03:
Some progress on UVC: https://github.com/zephyrproject-rtos/zephyr/discussions/63421#discussioncomment-10232463

mmahadevan108, 2024-08-13:
@jfischer-no , we have moved this to be in the 4.0 Release Plan for the conversion of the USB Device. 
Do you want to create a separate issue for the remaining tasks related to USB Host?

mmahadevan108, 2024-08-28:
@jfischer-no, can you please share some insight on what the plans are for the remaining tasks related to USB Host. Should we create a separate RFC for this so we can track it for future Zephyr releases.

jfischer-no, 2024-08-29:
> @jfischer-no, can you please share some insight on what the plans are for the remaining tasks related to USB Host. Should we create a separate RFC for this so we can track it for future Zephyr releases.

**No.**

As mentioned the beginning:
"This is a meta issue for maintainers and developers of relevant subsystems to track progress on new USB support."
This is not a release specific issue. There are parts of device support that depend on parts in host support. Please keep this topic free of irrelevant discussions, (also free of irrelevant concerns and wishes).

mmahadevan108, 2024-08-29:
@jfischer-no, we have moved this meta issue to be fixed in 4.0. How should we track the items that are not addressed in 4.0 release?

josuah, 2025-03-17:
I think this list was for the features that were present in `device` and their migration towards `devce_next`.

About UVC, the protocol supports output and input: a webcam can receive a video signal from the host even though that is rarely used.

This would allow to use https://github.com/zephyrproject-rtos/zephyr/pull/76798 with light modification rather than going the road of implementing host-side UVC in Zephyr, in case that solves your use-case.

MaheshAvula-Alifsemi, 2025-04-08:
Hi ,
Has anyone tried porting the TinyUSB host stack to Zephyr, or can someone provide some pointers on how to get started with porting TinyUSB host stack support in Zephyr?"

tannewt, 2025-04-08:
> Hi , Has anyone tried porting the TinyUSB host stack to Zephyr, or can someone provide some pointers on how to get started with porting TinyUSB host stack support in Zephyr?"

TinyUSB now supports Zephyr as an RTOS. I haven't tried the host stack yet but it does work for device. (We're using it in CircuitPython's Zephyr port.) https://github.com/hathach/tinyusb/pull/2962

MaheshAvula-Alifsemi, 2025-04-09:
Hi @tannewt , Thank you for the inputs, I will just go though it.

dleach02, 2025-07-01:
@jfischer-no I added USB Video class to the USB host section. We are going to work on this for a customer requirement we have with Zephyr.

josuah, 2025-07-21:
I updated USB Host with "USB Video class support (WIP @josuah)".