zephyrproject-rtos/zephyr #42072: west: spdx: Blank FileChecksum field for missing build file

swinslow, 2022-01-23:
**Describe the bug**

The `west spdx` functionality uses the output from CMake's file-based API to determine which build files are created from the various build targets. It then checks each built file to get its checksum and other details for the `build.spdx` SPDX Document.

Using the current main branch to build some basic samples (e.g. samples/hello_world, samples/basic/blinky), there is now a file which is being reported by the CMake metadata as being built: `build/zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a`.

However, when `west spdx` tries to gather this file's checksum after the build is complete, this file is not present. As a result, `west spdx` includes the following line:
```
FileChecksum: SHA1: 
```

which causes the SPDX Document to be invalid due to the missing checksum after `SHA`:`.

**To Reproduce**

Steps to reproduce the behavior:
1. run `west build` followed by `west spdx` as described in the [Zephyr documentation](https://docs.zephyrproject.org/latest/guides/west/zephyr-cmds.html#software-bill-of-materials-west-spdx)
2. try to [validate](https://tools.spdx.org/app/validate/) the contents of the `build.spdx` Document created by `west spdx`; it will fail due to the missing checksum.

Note also the separate issue at #42070 which may cause other validation warnings, and which is being addressed as well.

**Impact**

The SPDX Document being created is not a valid SPDX Document, and might not be consumable by downstream tools that expect full conformance to the spec.

**Environment (please complete the following information):**

- Linux (Ubuntu 20.04.3 LTS)
- Zephyr SDK 0.13.2
- Zephyr main branch, commit d51a67b7c0
  - with one not-yet-committed change to zspdx/writer.py for #42070, which should have no relevance to this bug

**Additional context**

I'm not familiar enough with the full Zephyr build process to understand why the `libdrivers__interrupt_controller.a` file is no longer present once the build is complete. I don't know whether that reflects a separate bug.

However, regardless of why that's occurring, `west spdx` should be able to gracefully handle errors such as this, and should not generate invalid SPDX Documents even if files are unexpectedly missing.

swinslow, 2022-01-23:
cc @kestewart for visibility

My current plan to address this is to modify the zspdx walker.py and/or scanner.py files so that a build file will be ignored if CMake reports it as being built, but it is not found on disk after the build is complete.

mbolivar-nordic, 2022-03-15:
@swinslow any plans to address this in the near future?

swinslow, 2022-03-19:
Hi @mbolivar-nordic, this should now be fixed with #43991. Non-existent files which were causing this bug will now be omitted from the SPDX document so that they will not cause it to be invalid due to no available checksum value.

@kestewart I added you as a second reviewer on that issue as well.

PR #43991 (2022-03-24): west: spdx: Exclude files not present after build
The list of files which are included in the `build.spdx` SPDX SBOM document
is based on the files recorded as build artifacts based on the CMake
file-based API metadata response.

In some situations, such as the case indicated in #42072, a build artifact
may be reported by CMake but no such file is present on the system
following the build. This results in the `build.spdx` SPDX SBOM being
invalid, as a result of trying to provide metadata for a non-existent
file (and specifically being unable to provide its checksum).

This commit fixes this bu [... cut]
commit c2ae5b3b (2022-03-24): west: spdx: Exclude files not present after build

The list of files which are included in the `build.spdx` SPDX SBOM document
is based on the files recorded as build artifacts based on the CMake
file-based API metadata response.

In some situations, such as the case indicated in #42072, a build artifact
may be reported by CMake but no such file is present on the system
following the build. This results in the `build.spdx` SPDX SBOM being
invalid, as a result of trying to provide metadata for a non-existent
file (and specifically being unable to provide its checksum).

This commit fixes this b [... cut]
closed 2022-03-24 by carlescufi