zephyrproject-rtos/zephyr #42082: Add intermediate cast to void* in CONTAINER_OF to make it forget about alignment.

mkschreder, 2022-01-24:
PC-Lint Plus identifies that CONTAINER_OF is messing with alignment:
warning 2445: cast from 'char *' to 'struct s2lp_emul *' increases required alignment from 1 to 8

The solution is to add an intermediate cast to void* in the definition of CONTAINER_OF in include/sys/util.h to make the compiler forget about alignment. 
`#define CONTAINER_OF(ptr, type, field) ((type *)(void*)(((char *)(ptr)) - offsetof(type, field)))`

bulanek, 2022-10-07:
Still not resolved. This is not only enhancement but it can hide errors which leads to read/write from unexpected memory part. Please make zephyr be ok with gcc flag -Werror=cast-align.

henrikbrixandersen, 2022-11-17:
Please submit a PR with a proposed fix.