zephyrproject-rtos/zephyr #42098: intel_adsp_cavs25: west sign command output some unrecognized ASCII char.

chen-png, 2022-01-24:
**Describe the bug**
for intel_adsp_cavs25 platform, when doing **west sign command, it reported some unrecognized ASCII char when parsing cse entry** name "cavs0015.met", the output as below, there are some unrecognized char after "cavs0015.met".
please ignore data of entry.offset and length, I removed them due to private policy.
```
 partition_name: 'ADSP'
         header_version: 2
          entry_version: 1
             nb_entries: 3
             entry.name: 'ADSP.man'
           entry.offset: 
           entry.length:
             entry.name: 'cavs0015.met▒'
           entry.offset: 
           entry.length: 
             entry.name: 'cavs0015'
           entry.offset: 
           entry.length: 
```
this will causing an **error "UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc0 in position 899: invalid start byte" when using twister** command, because twister needs to decode the output string of "west flash" command.
the raw data of "west flash" captured by twister at position 899 as below, cannot decode \xc0\x04 using utf-8.
``` 
entry.name: 'cavs0015.met\xc0\x04'\n
```

**Debug Info**
I found that in tgl.toml file, there is a cse entry as below, the strlen(name) is 12 bytes,
```
[[cse.entry]]
name = "cavs0015.met"
offset = 
length = 
```
but the **cse entry definition in rimage source code is as below, the entry name is uint8 [12]**, it includes the empty char for string. so when **rimage parsing entry name "cavs0015.met", it just overlay the empty char**, then when we output this entry_name, it will output some other chars until meeting another empty char.
**so when we define a cse entry name, the max strlen(name) should be 11 chars, not 12 chars.**
```
struct CsePartitionDirEntry {
        uint8_t  entry_name[12];
        uint32_t offset;
        uint32_t length;
        uint32_t reserved;
}  __attribute__((packed));
```

**Suggestions**
**Could we using another name for cse entry "cavs0015.met", less than 12 chars**.
I tested this method, it worked well.

**Environment (please complete the following information):**
 - Zephyr Commit:  d51a67b7c098c

chen-png, 2022-01-24:
hi @andyross could you help take a look of this?

andyross, 2022-01-24:
This is a bug against rimage, can you refile it there?  https://github.com/thesofproject/rimage

I don't know if it's OK to change the entry_name field or not.  That metadata isn't used by Zephyr, it's for parsing by a host audio driver.  I'd be afraid to touch it without coordination with SOF and the windows driver people, honestly.

FWIW, as I read it it's just a display bug.  There's nothing wrong with that 12-byte string, the bug is that rimage assume's that it's null-terminated in memory when the manifest format doesn't require it.

Also: a Zephyr bug I think you **should** file is that the binary garbage in the log string is the proximate cause of twister blowing up with a utf-8 decode exception.  Twister needs to be tolerant of tools that emit bad strings.

closed 2022-01-24 by andyross

chen-png, 2022-01-25:
> I read it it's just a display bug. There's nothing wrong with that 12-byte string

@andyross  yes, it is just a display bug and it is captured by twister just right. 
we can add a handler in twister to catch this exception when decoding output of west flash command, but it would be better to fix it from rimage if we can't modify entry name arbitrarily, i.e, increase the bytes in entry definition structure, I will report this in 
 sof rimage project too.

chen-png, 2022-01-25:
reported an issue in rimage project https://github.com/thesofproject/rimage/issues/86

PR #42114 (2022-01-25): twister: ignore unencodable unicode of west flash command.
sometimes there are some unencodable unicode chars from output
of west flash command, need to ignore them rather than reporting
an error.

this fix is from zephyr side, add handler to handle unencodable unicode chars in twister. 

fix: #42098