zephyrproject-rtos/zephyr #42102: doc: searches for sys_reboot() are inconsistant

pearson, 2022-01-24:
It is difficult to find documentation on `sys_reboot()` and associated symbols. Searching through the main Zephyr documentation site gives different search results from a search starting from the Zephyr API Documentation Files List.

Searching through the Zephyr documentation site:
- searching for `sys_reboot()` only lists a couple mentions of the function, as well as `CONFIG_REBOOT`, but no documentation on the function itself
- searching for `SYS_REBOOT_WARM` gives no results
- searching for `SYS_REBOOT_COLD` lists one result of the define in use, but no description of it

Searching from `[File List](https://docs.zephyrproject.org/latest/doxygen/html/files.html)` or other pages in the File List tree:
- typing `sys_reboot()`, `SYS_REBOOT_WARM`, or `SYS_REBOOT_COLD` into the search box shows a drop-down menu with a link to [reboot.h File Reference](https://docs.zephyrproject.org/latest/doxygen/html/reboot_8h.html). This page does show some minimal documentation for `sys_reboot()`.

My expectation would be to find search results for `sys_reboot()` when doing so from the main Zephyr documentation page.

manoj153, 2022-02-23:
Concur, I do not know WARM vs COLD.

closed 2022-05-09 by github-actions[bot]

owain-incus, 2022-08-08:
Kind of a pretty basic functionality one would want to use.
Should be documented.

bandogora, 2023-05-16:
Bumping this because I'm looking for documentation right now.

kartben, 2023-05-16:
Related to issue #55173.

kartben, 2023-07-31:
This will be fixed with #60920, courtesy of @gmarull ❤️