zephyrproject-rtos/zephyr #42108: upsquared: isr_dynamic test is failing

yerabolu, 2022-01-25:
Describe the bug
isr_dynamic test is failing on up_squared platform

What have you tried to diagnose or workaround this issue?

To Reproduce
Steps to reproduce the behavior:

twister -p up_squared --device-testing --device-serial /dev/ttyUSB1 -T tests/kernel/interrupt/ --west-flash="$HOME/up_squared_efi.sh" -j 1 -vv
See error
Expected behavior
test should pass.

Logs and console output
```
*** Booting Zephyr OS build v2.7.99-1382-g742412606090  ***
Running test suite interrupt_feature
===================================================================
START - test_isr_dynamic
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/interrupt/src/dynamic_isr.c:101: test_isr_dynamic: (x86_irq_funcs[TEST_IRQ_DYN_LINE] == dyn_isr && x86_irq_args[TEST_IRQ_DYN_LINE] == (void *)ISR_DYN_ARG is false)
dynamic isr did not install successfully
FAIL - test_isr_dynamic in 0.26 seconds
===================================================================
START - test_nested_isr
SKIP - test_nested_isr in 0.1 seconds
===================================================================
START - test_prevent_interruption
locking interrupts
unlocking interrupts
timer fired
PASS - test_prevent_interruption in 2.6 seconds
===================================================================
START - test_isr_regular
irq(27)=vector(64)
irq(28)=vector(65)
not testing irq enable/disable
PASS - test_isr_regular in 0.8 seconds
===================================================================
START - test_isr_offload_job_multiple
PASS - test_isr_offload_job_multiple in 0.1 seconds
===================================================================
START - test_isr_offload_job_identi
PASS - test_isr_offload_job_identi in 0.1 seconds
===================================================================
START - test_isr_offload_job
irq(49)
PASS - test_isr_offload_job in 0.1 seconds
===================================================================
START - test_direct_interrupt
SKIP - test_direct_interrupt in 0.1 seconds
===================================================================
Test suite interrupt_feature failed.
===================================================================
PROJECT EXECUTION FAILED
```

Environment:

OS: Fedora
Toolchain: SDK 0.13.2
Commit SHA: d51a67b7c098

yerabolu, 2022-01-25:
Tests are also failing on EHL_CRB

enjiamai, 2022-02-10:
This issue was caught when my another interrupt testing PR got merged. I change the way how we trigger the interrupt (FROM INT instruction to send IPI).  I found the trigger the an interrupt by IPI to CPU core itself doesn't work on ehl_crb, both in xapic and x2apic mode. Still in progress.

PR #43254 (2022-03-09): tests: testsuite: add emulating send IPI in X2APIC mode
We emulate the interrupt by sending the IPI to core itself by
the local APIC for x86 platfrom.

But in X2APIC mode, this no longer works. So we emulate the
interrupt the by writing the IA32_X2APIC_SELF_IPI MSR to send
IPI to the core itself via LOAPIC also. According to SDM vol.3
chapter 10.12.11.

Fixes #42108
commit 2b1bccde (2022-03-09): tests: testsuite: add emulating send IPI in X2APIC mode

We emulate the interrupt by sending the IPI to core itself by
the local APIC for x86 platfrom.

But in X2APIC mode, this no longer works. So we emulate the
interrupt the by writing the IA32_X2APIC_SELF_IPI MSR to send
IPI to the core itself via LOAPIC also. According to SDM vol.3
chapter 10.12.11.

Fixes #42108
closed 2022-03-09 by nashif