zephyrproject-rtos/zephyr #42112: OTS: L2CAP: Unable to find channel of LE Credits packet

asbjornsabo, 2022-01-25:
Running the mcs_mcc BabbleSim tests, the following error message occurs when transferring objects over L2CAP:

d_01: @00:00:14.031757  [00:00:14.031,738] <dbg> bt_mpl: on_object_send: Read complete
d_01: @00:00:14.031757  [00:00:14.031,738] <dbg> bt_ots: l2cap_disconnected: Channel 0x82523fc disconnected
d_01: @00:00:14.032209  [00:00:14.032,196] <err> bt_l2cap: Unable to find channel of LE Credits packet
d_00: @00:00:14.032467  [00:00:14.032,440] <dbg> bt_ots: l2cap_disconnected: Channel 0x82549a8 disconnected
d_00: @00:00:14.032467  [00:00:14.032,440] <dbg> bt_otc: closed: L2CAP closed, context: 0x82549a4, conn: 0x825f0c0

carlescufi, 2022-01-27:
@sjanc does this ring a bell at all?

asbjornsabo, 2022-02-02:
I plan to look at this in the sprint that starts today.

PR #42445 (2022-02-09): Bluetooth: Host: L2CAP: Don't send credits if chan is disconnected
Fixes bug where Host sends L2CAP Flow Control Credit after receiving a
L2CAP Disconnect Response.

The when the callback in l2cap_chan_le_recv_sdu is handed a Disconnect
Reponse, the connnection becomes disconnected, as illustrated by the
asserts. The function should now trigger sending credits if the
connection is disconnected after the callback returns.

Fixes #42112
commit ae9380d9 (2022-02-09): Bluetooth: Host: L2CAP: Don't send credits if chan is disconnected

Fixes bug where Host sends L2CAP Flow Control Credit after receiving a
L2CAP Disconnect Response.

The when the callback in l2cap_chan_le_recv_sdu is handed a Disconnect
Reponse, the connnection becomes disconnected, as illustrated by the
asserts. The function should now trigger sending credits if the
connection is disconnected after the callback returns.

Fixes #42112
closed 2022-02-09 by carlescufi