zephyrproject-rtos/zephyr #42113: Modbus RTU driver reports frame length error

mjm987, 2022-01-25:
The Modbus RTU driver (modbus_serial.c) hard-codes the nr of stopbits for the serial line to 1 if using parity or 2 without parity.
This corresponds to the Modbus RTU specification but there exists sensor devices which sends non-conform RTU frames with one stopbit and no parity.
In this case the modbus rtu driver reports a "frame length error" without delivering the data - at least on the nrf52 platform.

Background:
 - If a nrf52840 is configured to 2 stopbits, it not only sends 2 stopbits but it also expects receiving two stopbits. Effectly it is a parity set to mark which is a hardware restriction.
 - When hardcoding modbus_serial.c to one stopbit, this specific (misbehaved) sensor works.

**Expected behavior**
The nr of stopbits (1 or 2) should be a modbus config parameter which defaults to the modbus rtu specification when not defined (0).

**Impact**
None

**Logs and console output**
[10:22:29.106,384] <wrn> modbus_serial: Frame length error

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit: c71dd808345b33b005a9cbba0a1f22edec729daf

jfischer-no, 2022-01-25:
>This corresponds to the Modbus RTU specification but there exists sensor devices which sends non-conform RTU frames with one stopbit and no parity.

This is not a bug because, as you write, the frames are not spec compliant and therefore your sensor is not a MODBUS RTU device. There is https://github.com/zephyrproject-rtos/zephyr/pull/40558 to allow this kind of frames but that should be limited to client configuration.

mjm987, 2022-01-25:
Ok, but adding a parameter eg. "nr_stopbits" to "struct modbus_iface_param" which defaults (when 0) to the RTU standard would solve the problem for modbus client as well for server configurations without side effects (or not?)

chrta, 2022-01-25:
This is related to https://github.com/zephyrproject-rtos/zephyr/pull/40558 We will update the pr in a few weeks.

closed 2022-03-24 by nashif