zephyrproject-rtos/zephyr #42122: LoRa Driver extended features

mjm987, 2022-01-25:
**Is your enhancement proposal related to a problem? Please describe.**
The current implementation of the LoRa driver seems to be based on the LoRaWAN default configuration but for standalone LoRa applications, some features are missing, eg. carrier activity detection (CAD), CRC on/off, fixed length payload (Header suppression), IQ inversion, paDutyCycle, etc. etc.

**Describe the solution you'd like**
Implementation of all possible features

**Describe alternatives you've considered**
Have a software interface in the LoRa driver that allows calling arbitrary sx12xx commands via the lora radio library.

5frank, 2022-09-09:
Other suggested changes to `struct lora_modem_config`:
- `datarate` is arguably incorrect name as it is currently expected to be a spreading_factor value. Actual data rate in bit/s is given by:  `datarate = spreading_factor / symbol_period`. The term used in this way only in Semtech LoRaMac-node code but not in Semtech sx12xx reference manuals. Confusing it is...

- Change bandwith to be in hertz not enum. (example BW_1625_KHZ is missing)

- enums should have a namespace prefix `LORA_`.
- Remove `bool tx` in config? Perhaps I am missing something, but calling lora_config twice for tx and rx doesn't seem right?

I will try to make a PR when I have time but I'm afraid lots of things would break in this fork https://github.com/zephyrproject-rtos/loramac-node.

Example on some changes to lora.h: 
```C
struct lora_coding_rate {
    uint8_t info_bits : 3;  //!< useful information bits
    uint8_t data_bits : 4;  //!< total data bits
    uint8_t interleave : 1; //!< long interleaving if set
};

#define LORA_CR(INFO_BITS, DATA_BITS, INTERLEAVE)                              \
    (struct lora_coding_rate){                                                 \
        .info_bits = (INFO_BITS),                                              \
        .data_bits = (DATA_BITS),                                              \
        .interleave = (INTERLEAVE)                                             \
    }

#define LORA_CR_4_5   LORA_CR(4, 5, 0)
#define LORA_CR_4_6   LORA_CR(4, 6, 0)
#define LORA_CR_4_7   LORA_CR(4, 7, 0)
#define LORA_CR_4_8   LORA_CR(4, 8, 0)

struct lora_modem_config {
    uint32_t frequency; //!< frequency in Hz
    uint32_t bandwidth; //!< bandwidth in Hz
    uint16_t preamble_len;
    struct lora_coding_rate coding_rate;
    uint8_t spreading_factor;
    int8_t tx_power; //!< transmit power in dBm
};

/**
 * @return data rate in bits per second (bps) */
static inline uint32_t lora_datarate_bps(const struct lora_modem_config *config)
{
    uint32_t sf = config->spreading_factor;
    uint32_t bw = config->bandwidth;

    // avoid div by zero below
    if (sf >= 32) {
        return 0;
    }
    /* pow2(sf) */
    uint32_t pow2_sf = (uint32_t)1 << sf;

    /* symbol period length in seconds  calculated as:
     *    uint32_t t_symbol =  pow2_sf / config->bandwidth; */

    /* div round near */
    uint32_t datarate_bps = (sf * bw + (pow2_sf -1)) / pow2_sf;

    return datarate_bps;
}
```

warrenb78, 2023-09-18:
Another big issue I have with this driver is the lack of support of all bandwidths, which is cuurently limited to 125, 250, 500 KHz.