zephyrproject-rtos/zephyr #42130: drivers: serial: remove struct uart_device_config

gmarull, 2022-01-25:
## Introduction

I've opened this RFC to question if `struct uart_device_config` should be removed.

### Problem description

The main reasons:

1. It is not used by all drivers (in fact, I think it's a minority using it)
2. It pretends to be generic, but it is not (see for instance)
3. No other subsystems are using this kind of _inheritance_ mechanism.

### Proposed change

Each driver is responsible to store whatever is needed.

## Detailed RFC

See above.

### Proposed change (Detailed)

1. Remove usage of `struct uart_device_config` from all drivers
2. Remove `struct uart_device_config`

### Dependencies

None

### Concerns and Unresolved Questions

None

## Alternatives

N/A

gmarull, 2022-01-25:
cc: @dcpleung

dcpleung, 2022-01-25:
Probably a good move since only a few drivers are using it.

PR #42131 (2022-02-22): drivers: serial: drop uart_device_config
This PR removes `struct uart_device_config` from drivers using it, and does some cleanups thanks to this removal. The main points to remove this structure:

1. It is not used by all drivers (in fact, I think it's a minority using it)
2. It pretends to be generic, but it is not. For example, the field that stores the register base address is not suitable for most drivers.
3. No other subsystems are using this kind of inheritance mechanism (I think).
4. Some fields are not always used, wasting extra ROM space (e.g. sys_clk_freq).

Fixes #42130
closed 2022-02-22 by nashif