zephyrproject-rtos/zephyr #42134: TLS handshake error using DTLS on updatehub

youssefchei-xite, 2022-01-25:
**Describe the bug**
I'm testing the **updatehub** using nrf5340dk and quectel BG95 as modem.
I used the zephyr/samples/subsys/mgmt/updatehub sample and I used the **updatehub cloud** (free version for now) for update 
1- I build and flash version 1.0.0
2-Create and push the update package 2.0.0 to updatehub cloud using https://docs.updatehub.io/zephyr-project/zephyr-project-guide/ , steps: 8-9 then create rollout using step 11
3- start the FOTA using **updatehub run** command
4- the nrf5340dk connect to **updatehub cloud**, download the binary and update it after reboot and the firmware 2.0.0 is running.

Then I tested the updatehub with DTLS by including overlay-dtls.conf using the default certificates.
But I always got
[00:02:15.450,164] <err> net_sock_tls: TLS handshake error: -6800
[00:02:15.459,075] <err> updatehub: Could not send request
[00:02:15.473,327] <err> main: Invalid response

Is there a certificate that must upload to updatehub cloud?
Is there any config missing to make the updatehub example connect to updatehub cloud using DTLS ?

**To Reproduce**
Steps to reproduce the behavior:

build the updatehub sample
west build --board=nrf5340dk_nrf5340_cpuapp -d build zephyr/samples/subsys/mgmt/updatehub --\
-DOVERLAY_CONFIG="overlay-modem.conf;overlay-dtls.conf"

flash the binaries
west flash -d build

**Logs and console output**
```
[00:00:11.155,303] <inf> main: Network connected
[00:00:12.205,657] <inf> main: Starting UpdateHub manual mode
uart:~$
[00:02:15.450,164] <err> net_sock_tls: TLS handshake error: -6800
[00:02:15.459,075] <err> updatehub: Could not send request ret:-1 errno:0x71  113
[00:02:15.473,327] <err> main: Invalid response

```
**Environment (please complete the following information):**

- OS: MacOS
- Toolchain :nrf Connect SDK v1.7.0
- Zephyr version used:v2.6.99-ncs1

rlubos, 2022-01-26:
I'm not that familiar with updatehub, but I'd check with `updatehub cloud` support first if their DTLS server is active at all - I've tried to establish DTLS connection with their server (`coap.updatehub.io:5684`) with both, Zephyr (mbedTLS) and openSSL, and it's simply unresponsive (no reply for the initial Client Hello at all), just as if the server was not running under the secure port.

boaks, 2022-01-27:
Using GET coap://coap.updatehub.io/ (without DTLS) results in 2.05 (and sporadically in 5.03). I guess, you need to contact the operator.

carlescufi, 2022-02-02:
@youssefchei-xite can you please respond to the comments?

youssefchei-xite, 2022-02-02:
@rlubos 
Did you get an answer from updatehub cloud support about the DTLS server?
@boaks 
Updatehub cloud without DTLS works fine, I didn't understand which operator I should contact

rlubos, 2022-02-02:
@youssefchei-xite No, I did not contact updatehub - I'd rather expect that you do it as a issue reported and a party interested in resolving the issue. You should not expect Zephyr maintainers to resolve problems with 3rd party software.

boaks, 2022-02-02:
>  I'd rather expect that you do it 

That was also my consideration.
I don't know, who's operating updatehub, but it feels strange for me, that you hand over firmware to someone, you don't know or have a contract with. I guess, if you have registered there, so you should have also a way to contact them.

rlubos, 2022-02-02:
@nandojve @otavio Do you know perhaps if DTLS port is active for updatehub cloud?

nandojve, 2022-02-02:
Hi @youssefchei-xite ,

> I didn't understand which operator I should contact

You need to talk with @otavio about UpdateHub Cloud services.

carlescufi, 2022-02-09:
@youssefchei-xite @nandojve ping, is this resolved?

nandojve, 2022-02-09:
Hi folks,

I'm still waiting UpdateHub Cloud team. It seems some people are in vacations and that create some delays.
What I can say right now is, I made a pretty simple test using libCoAP tools and confirmed that UpdateHub Cloud API allows connection with both DTLS and non-DTLS using as backend the GnuTLS library. However, that is not enough to send a well defined example to @rlubos, which could definitely help us to understand root of issue.

CC: @otavio @gustavosbarreto

boaks, 2022-04-11:
> libCoAP ... GnuTLS

With that I was also able to execute an handshake, my findings:

It seems to support only RSA based cipher suites, no ECDSA (**NOT** compliant to [RFC7252, 9.1.3.3](https://datatracker.ietf.org/doc/html/rfc7252#section-9.1.3.3) ).
The server's certificate has expired on "Wed Mar 18 15:19:22 CET 2020", it's a selfsigned one, and has no subject to verify.

So, for an "successful" handshake, you may need to disable some of the common checks.
Not clear, if this is a good idea and considered by that service.

@youssefchei-xite 

Anyway, I think, if you still on it, you may provide a update from your side. 
If you gave up, you close this issue.

closed 2022-06-25 by github-actions[bot]