zephyrproject-rtos/zephyr #42136: doc: Create documentation for all shell commands.

keith-zephyr, 2022-01-25:
**Is your enhancement proposal related to a problem? Please describe.**
Capturing a discussion started on [discord](https://discord.com/channels/720317445772017664/884782246010175528/931473463535874108)

Users cannot easily discover what shell commands are available, and the built-in help for shell commands doesn't adequately describe their usage.  Users often have to go to the shell command source to determine correct usage.

**Describe the solution you'd like**
Shell documentation should meet these requirements:
- Auto-generated from the source
- Tagged so that shell commands can be included with the module/driver that implements the command, and in a central location documenting all shell commands.
- Standardized usage text such as `<name> <command> [<subcommand> ...] [<arg0> ...]` - suggested by marinjurjevic

**Potential Technologies to leverage**
-  [Python argparse](https://docs.python.org/3/library/argparse.html), [C implementation](https://github.com/cofyc/argparse.) - suggested by Emil Gydesen
- Auto-generate the shell source code from a YAML file describing shell commands  - suggested by Henrik Brix Anderson

carlescufi, 2022-01-25:
@jakub-uC do you have suggestions on how to achieve this?

gregshue, 2022-01-25:
> Tagged so that shell commands can be included with the module/driver that implements the command

Just to check my understanding ... you want the document generation process to extract shell command documentation from out-of-tree Zephyr modules and put them into an aggregate list (like already happens for Kconfig and interface documentation)?

If so, I agree.

gregshue, 2022-01-25:
BTW, the shell command names need to begin with a prefix indicating the originating module to avoid name collisions.

> and in a central location documenting all shell commands.

Is this documentation for what exists in the workspace, or what is built into the executable?

jakub-uC, 2022-01-26:
I would like to understand a little more about the `argparse` implementation. What specifically, would it look like? If we use it for subcommand then in my opinion we will lose :

- developing subcommands using the `tab` button. 
- support for dynamic commands. At least for now, I can't think of a way to implement this.
- narrowing the command tree with the `select` command. This is widely used when implementing `AT` commands.

If we want to have a list of all the commands that are in the shell, it seems to me that the easiest solution would be to add code that generates documentation to the macros that create the commands. 

Once again, I do not want to defend the current solution at all costs, but it is consistent and takes into account many use cases.

keith-zephyr, 2022-01-26:
> BTW, the shell command names need to begin with a prefix indicating the originating module to avoid name collisions.
> 
> > and in a central location documenting all shell commands.
> 
> Is this documentation for what exists in the workspace, or what is built into the executable?

My ask is to document the shell commands built into the executable.