zephyrproject-rtos/zephyr #42164: Bluetooth tester application is broken after switch to logging v2

sjanc, 2022-01-26:
**Describe the bug**

Bluetooth tester application is crashing on start.

git bisect results in following commit:

262cc55609b73ea61b5f999c6c6daaba20bc5240 is the first bad commit
commit 262cc55609b73ea61b5f999c6c6daaba20bc5240
Author: Krzysztof Chruscinski <krzysztof.chruscinski@nordicsemi.no>
Date:   Wed Jan 12 15:52:48 2022 +0100

    logging: Deprecate v1, default to v2
    
    Reduced logging mode selection to deferred, immediate, minimal and
    frontend. Decoupled logging version from mode and created CONFIG_LOG1
    which can be used to explicitly select deprecated version.
    
    From now on, chosing CONFIG_LOG_MODE_{IMMEDIATE,DEFERRED} will result
    in version2.
    
    Deprecated CONFIG_LOG2_MODE_{IMMEDIATE,DEFERRED} with cmake warning.
    
    Codebase adapted to those changes.
    
    Signed-off-by: Krzysztof Chruscinski <krzysztof.chruscinski@nordicsemi.no>

In debugger we see:
Debugger was unable to disassemble function containing address 0x00000000deadbeee: Cannot access memory at address 0xdeadbaee

We don't even get into tester thread, when stepping manually we get k_thread_abort right after returning from tester_init() function.

gdb) backtrace
#0  k_thread_abort (thread=0x200038e0 <z_main_thread>) at zephyrproject/build/zephyr/include/generated/syscalls/kernel.h:206
#1  z_thread_entry (entry=0x3f715 <bg_thread_main>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at zephyrproject/zephyr/lib/os/thread_entry.c:38
#2  0x0000c540 in arch_switch_to_main_thread (main_thread=<optimized out>, stack_ptr=0x2000a440 <z_idle_stacks> "K\255\271\317\036<\225\005\324ͽ}\022W0\031\373\375<\314\001", _main=0x3f715 <bg_thread_main>) at zephyrproject/zephyr/arch/arm/core/aarch32/thread.c:557
#3  0x05953c1e in ?? ()

We use nRF52840-DK.

A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nordic-krch, 2022-01-26:
Strange. I run it, added additional log message to `cmd_handler` in bttester.c and got it out on RTT. Can you try to add `CONFIG_LOG1=y` to your configuration and check again? `CONFIG_LOG1=y` should fall back to the state before that commit.

sjanc, 2022-01-27:
OK, so crashing on start was mistake from my side. Autopts is restarting board and debugger was going crazy. So there is no crash on init, but tests are not passing with V2 logs (not even a single one). We see some BTP communication which seems to be OK, yet tests are timing out.

Setting CONFIG_LOG1=y fix the issue and tests are passing...

We will try to debug it further but it does look like V2 logs are messing something for tester

nordic-krch, 2022-01-27:
There is a slight performance degradation compared to v1. Another thing could be stack overflow due to increased stack usage when log2 is used (i have no number how they compare). Can you try disable `BT_DEBUG` to check if it is in bluetooth debugging?

sjanc, 2022-01-28:
Hi,

disabling BT_DEBUG doesn't fix the issue

sjanc, 2022-01-28:
fixed with https://github.com/zephyrproject-rtos/zephyr/commit/a40ca6fd1c8b730fa372768d4d14d601785874e2

closed 2022-01-28 by sjanc