zephyrproject-rtos/zephyr #42168: v3.0.0-rc1: mimxrt685_evk_cm33: i2s driver build failure

hakehuang, 2022-01-26:
**Describe the bug**
i2s driver build failure driver .

Please also mention any information which could help others to understand
the problem you're facing:
build failure 

 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make

**Expected behavior**
Buiid PASS

**Impact**
I2S driver

**Logs and console output**

```
/home/ubuntu/zephyr_project/zephyr_rt685/zephyr/drivers/dma/dma_mcux_lpc.c: In function 'dma_mcux_lpc_reload':
/home/ubuntu/zephyr_project/zephyr_rt685/zephyr/drivers/dma/dma_mcux_lpc.c:383:29: error: 'data' undeclared (first use in this function)
  383 |  uint32_t virtual_channel = data->channel_index[channel];
      |                             ^~~~
/home/ubuntu/zephyr_project/zephyr_rt685/zephyr/drivers/dma/dma_mcux_lpc.c:383:29: note: each undeclared identifier is reported only once for each function it appears in
/home/ubuntu/zephyr_project/zephyr_rt685/zephyr/drivers/dma/dma_mcux_lpc.c:382:32: warning: unused variable 'dev_data' [-Wunused-variable]
  382 |  struct dma_mcux_lpc_dma_data *dev_data = dev->data;
      |                                ^~~~~~~~
[119/200] Building C object zephyr/drivers/i2s/CMakeFiles/drivers__i2s.dir/i2s_mcux_flexcomm.c.obj

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, , )
 - Toolchain (e.g Zephyr SDK, ...): 0.13.0
 - Commit SHA or Version used: v3.0.0-Rc1

dleach02, 2022-01-26:
This is a result of something done globally last week ... has come up in a number of places.

dleach02, 2022-01-26:
duplicate of #42170

closed 2022-01-26 by dleach02