zephyrproject-rtos/zephyr #42170: v3.0.0-rc1: mimxrt685_evk_cm33: dma driver build failure

hakehuang, 2022-01-26:
**Describe the bug**
dma driver build failure driver .

Pease also mention any information which could help others to understand
the problem you're facing:
build failure

To Reproduce
Steps to reproduce the behavior:

mkdir build; cd build
cmake -DBOARD=mimxrt685_evk_cm33
make
Expected behavior
Buiid PASS

Impact
DMA driver

**Logs and console output**

```
/home/ubuntu/zephyr_project/zephyr_rt685/zephyr/drivers/dma/dma_mcux_lpc.c: In function 'dma_mcux_lpc_reload':
/home/ubuntu/zephyr_project/zephyr_rt685/zephyr/drivers/dma/dma_mcux_lpc.c:383:29: error: 'data' undeclared (first use in this function)
  383 |  uint32_t virtual_channel = data->channel_index[channel];
      |                             ^~~~
/home/ubuntu/zephyr_project/zephyr_rt685/zephyr/drivers/dma/dma_mcux_lpc.c:383:29: note: each undeclared identifier is reported only once for each function it appears in
/home/ubuntu/zephyr_project/zephyr_rt685/zephyr/drivers/dma/dma_mcux_lpc.c:382:32: warning: unused variable 'dev_data' [-Wunused-variable]
  382 |  struct dma_mcux_lpc_dma_data *dev_data = dev->data;
      |                                ^~~~~~~~
[114/168] Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/devices/MIMXRT685S/drivers/fsl_power.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/ubuntu/.local/bin/cmake --build /home/ubuntu/zephyr_project/zephyr_rt685/zephyr/tests/drivers/dma/loop_transfer/build
```

OS: (e.g. Linux, , )
Toolchain (e.g Zephyr SDK, ...): 0.13.0
Commit SHA or Version used: v3.0.0-Rc1

dleach02, 2022-01-26:
This is a result of a global change that occurred last week.

dleach02, 2022-01-26:
fixed by #42141

closed 2022-01-26 by dleach02