zephyrproject-rtos/zephyr #42177: PM_STATE_INFO_DT_ITEMS_LIST macro does not fill the pm_min_residency array

ctuzane, 2022-01-26:
I have assigned the cpu-power-states property via overlay as stated in the example device tree fragment. However, the PM_STATE_INFO_DT_ITEMS_LIST macro does not fill the pm_min_residency array in the policy_residency.c file. It is used as stated in the macro brief.

I am using the power management subsystem in residency mode. When I compile the code, I don't get any errors. In debug mode, I see that the pm_min_residency array is not filled in the pm_policy_next_state function.

I did not make any changes to the policy_residency.c file.

**Target Device:** nrf52833dk_nrf52833

### Overlay file:

    cpus {
        cpu@0 {
            cpu-power-states = <&state0 &state1>;
        };
    };
    
    power-states {
        state0: state0 {
            compatible = "zephyr,power-state";
            substate-id = <0>;
            power-state-name = "active";
            min-residency-us = <1000>;
            exit-latency-us = <100>;
        };

        state1: state1 {
            compatible = "zephyr,power-state";
            substate-id = <1>;
            power-state-name = "runtime-idle";
            min-residency-us = <10000>;
            exit-latency-us = <100>;
        };
    };

ctuzane, 2022-01-26:
I'm using the nrf system_off sample. In addition to this sample, I added cpu power states via overlay. I was expecting a power state to be selected according to the tick count of the next thread in the pm_policy_next_state function, but I saw that there was a problem and the active state was always selected.
    
    static volatile const struct pm_state_info pm_min_residency[] =
	PM_STATE_INFO_DT_ITEMS_LIST(DT_NODELABEL(cpu0));

 Then I put a breakpoint at the beginning of the pm_policy_next_state function and saw that the pm_min_residency array was not filled.

I am using Zephyr in Visual Studio Code via nRF Connect SDK. While creating the build configuration via nrf connect in Visual studio, I only gave the DTC_OVERLAY_FILE parameter. I don't have a problem during compilation, I have a problem with the macro filling the array.

**Expected behavior:** I was expecting the PM_STATE_INFO_DT_ITEMS_LIST macro in the policy_residency.c file to fill the processor power state definitions in the Devicetree into the pm_min_residency array.

**OS:** Windows
**Zephyr Version:** 2.6.99
**NCS Version:** 1.7.1

gmarull, 2022-01-26:
The code responsible for this part has been changed since the Zephyr version you report. There's a test that covers this part (https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/subsys/pm/policy_api/src/main.c). Can you try using upstream Zephyr?

ctuzane, 2022-01-27:
Thanks for your answer. When I examined zephyr v.3.0.0 in the Github repo, I saw that there was a change as you stated. But since this version is Pre-release yet, I will wait for it to be realese. I will let you know if there is a problem in my attempt after it is released.

closed 2022-01-27 by ctuzane