zephyrproject-rtos/zephyr #42181: Ethernet PHY imxrt1060 Teensy not initialising, sample with DHCPv4_client fails

EmbedWise, 2022-01-26:
**Describe the bug**
Ethernet PHY imxrt1060 Teensy not initialising, sample with DHCPv4_client fails. It does not get an IP and the green activity light is not showing any activity.

Please also mention any information which could help others to understand
the problem you're facing:
 - Target Platform TEENSY41
 - Have verified that the hardware is successfully working with TeensyDuino (FNET)

**To Reproduce**
Steps to reproduce the behavior:
1. Used branch 2.7.1 
2. Added this to the 
`SYS_INIT(teensy4_phy_reset, POST_KERNEL, CONFIG_PHY_INIT_PRIORITY);`
@MaureenHelm with issue #38473
3. Added the necessary changes to the prj.conf for USB support and SHELL and CONFIG_NET_L2_ETHERNET=y

```
# GENERAL
CONFIG_CODE_DATA_RELOCATION=y
CONFIG_LOG=y
CONFIG_GPIO=y

# USB (incl. console output)
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_UART_CONSOLE=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_DEVICE_PRODUCT="TEENSY USB CONSOLE"
CONFIG_USB_DEVICE_VID=0x2FE4
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y

CONFIG_SERIAL=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_UART_LINE_CTRL=y

# ETH
CONFIG_NETWORKING=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_ARP=y
CONFIG_NET_UDP=y
CONFIG_NET_DHCPV4=y

CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_INIT_STACKS=y

CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

CONFIG_NET_LOG=y
CONFIG_LOG=y

CONFIG_SLIP_STATISTICS=n
CONFIG_NET_SHELL=y

CONFIG_NET_L2_ETHERNET=y

# For using the shell
CONFIG_SHELL_PROMPT_UART="teensy:~$ "
CONFIG_PRINTK=y
CONFIG_SHELL=y
# Priority for backend is default 0 CDC ACM device initializes at POST_KERNEL at CONFIG_SHELL_BACKEND_SERIAL_INIT_PRIORITY which defaults to 50.
CONFIG_SHELL_BACKEND_SERIAL_INIT_PRIORITY=51
CONFIG_SHELL_BACKEND_SERIAL=y
CONFIG_THREAD_MONITOR=y
CONFIG_THREAD_NAME=y
CONFIG_SHELL_HISTORY=y
CONFIG_SHELL_STACK_SIZE=2048
CONFIG_SHELL_CMD_BUFF_SIZE=128
CONFIG_SHELL_VT100_COLORS=y
CONFIG_SHELL_STATS=y
CONFIG_SHELL_CMDS=y
CONFIG_SHELL_HELP=y
CONFIG_SHELL_TAB=y
CONFIG_SHELL_LOG_BACKEND=y
CONFIG_KERNEL_SHELL=y
CONFIG_DEVICE_SHELL=y
CONFIG_SENSOR_SHELL=n
CONFIG_SHELL_CMDS_SELECT=y
```

CMakeLists.txt
```
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.20.0)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(dhcpv4_client)

FILE(GLOB app_sources src/*.c)
target_sources(app PRIVATE ${app_sources})

zephyr_code_relocate(src/*.c DTCM_DATA)
zephyr_code_relocate(src/*.c ITCM_TEXT)
```
Note that I had to add **CONFIG_NET_L2_ETHERNET=y**

4. After booting I connect with USB and and send the command `net iface up 1`

**Expected behavior**
I would expect the device to try to get a DHCP lease. Which is unsuccessful. 

**Impact**
We want to use the RT1060 for our platform and therefore this is currently a show stopper

**Logs and console output**
```
[00:00:00.288,000] <inf> usb_cdc_acm: Device configured
[00:00:01.001,000] <wrn> net_if: iface 0x20202a98 is down
teensy:~$ net iface up 1
Interface 1 is up
[00:00:03.001,000] <wrn> net_if: iface 0x20202a98 is down
teensy:~$ net iface show

Interface 0x20202a98 (Ethernet) [1]
===================================
Link addr : 02:04:9F:B1:23:5A
MTU       : 1500
Flags     : NO_AUTO_START,IPv4
Ethernet capabilities supported:
        Virtual LAN
        Auto negotiation
        10 Mbits
        100 Mbits
IPv4 unicast addresses (max 1):
        <none>
IPv4 multicast addresses (max 1):
        <none>
DHCPv4 lease time : 0
DHCPv4 renew time : 0
DHCPv4 server     : 0.0.0.0
DHCPv4 requested  : 0.0.0.0
DHCPv4 state      : selecting
DHCPv4 attempts   : 0
```

**Environment (please complete the following information):**
 - OS:  MacOS
 - Toolchain gcc 
 - Commit tag 2.7.1

[dhcpv4_client_adapted.zip](https://github.com/zephyrproject-rtos/zephyr/files/7942698/dhcpv4_client_adapted.zip)

EmbedWise, 2022-01-28:
Finally got it to work on Zephyr 2.7.1 branch. With two steps:

- copy the changes from https://github.com/elcritch/zephyr/commit/55a3100b6eae952db37f6a814ee59303ebf0d661
- and adding using the following prj.conf:

```
# GENERAL
CONFIG_CODE_DATA_RELOCATION=y
CONFIG_LOG=y
CONFIG_GPIO=y

# SDHC
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_SDMMC_USDHC=y

# USB (incl. console output)
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_UART_CONSOLE=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_DEVICE_PRODUCT="TEENSY USB CONSOLE"
CONFIG_USB_DEVICE_VID=0x2FE4
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y

CONFIG_SERIAL=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_UART_LINE_CTRL=y

# ETH
CONFIG_NETWORKING=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=y

CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

CONFIG_NET_L2_ETHERNET=y
CONFIG_MDIO=y
CONFIG_MDIO_NXP_MCUX=y

CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=2
CONFIG_NET_IF_MCAST_IPV4_ADDR_COUNT=2
CONFIG_NET_IF_MAX_IPV4_COUNT=10
CONFIG_NET_IPV4_AUTO=y
# # CONFIG_NET_IPV4_LOG_LEVEL_DBG=y
# # CONFIG_NET_IPV4_AUTO_LOG_LEVEL_DBG=y
# CONFIG_NET_ICMPV4_LOG_LEVEL_DBG=y
# CONFIG_NET_DHCPV4_LOG_LEVEL_DBG=y
CONFIG_NET_DHCPV4_INITIAL_DELAY_MAX=10
CONFIG_NET_STATISTICS=y

CONFIG_NET_ARP=y
CONFIG_NET_SHELL=y
CONFIG_NET_CONFIG_AUTO_INIT=y

CONFIG_NET_LOG=y

# CONFIG_NET_CONFIG_SETTINGS=n
# CONFIG_NET_CONFIG_NEED_IPV6=n
# CONFIG_NET_CONFIG_NEED_IPV4=y
# CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.179.233"
# CONFIG_NET_CONFIG_MY_IPV4_NETMASK="255.255.255.0"

# SPI
CONFIG_SPI=y

# Floating Point Options
CONFIG_FPU=y
CONFIG_FPU_SHARING=y

# enable POSIX date and time api
CONFIG_POSIX_CLOCK=y

# enable the function sys_reboot()
CONFIG_REBOOT=y

# For using the shell
CONFIG_SHELL_PROMPT_UART="teensy:~$ "
CONFIG_PRINTK=y
CONFIG_SHELL=y
# Priority for backend is default 0 CDC ACM device initializes at POST_KERNEL at CONFIG_SHELL_BACKEND_SERIAL_INIT_PRIORITY which defaults to 50.
CONFIG_SHELL_BACKEND_SERIAL_INIT_PRIORITY=51
CONFIG_SHELL_BACKEND_SERIAL=y
# # CONFIG_OBJECT_TRACING=y
CONFIG_THREAD_MONITOR=y
# CONFIG_INIT_STACKS=y
# CONFIG_BOOT_BANNER=n
CONFIG_THREAD_NAME=y
CONFIG_SHELL_HISTORY=y
CONFIG_SHELL_STACK_SIZE=2048
CONFIG_SHELL_CMD_BUFF_SIZE=128
# CONFIG_SHELL_WILDCARD=y
# # CONFIG_SHELL_DYNAMIC_CMDS=n
CONFIG_SHELL_VT100_COLORS=y
# CONFIG_SHELL_HELP_ON_WRONG_ARGUMENT_COUNT=y
CONFIG_SHELL_STATS=y
CONFIG_SHELL_CMDS=y
CONFIG_SHELL_HELP=y
CONFIG_SHELL_TAB=y
# CONFIG_SHELL_METAKEYS=y
CONFIG_SHELL_LOG_BACKEND=y
CONFIG_KERNEL_SHELL=y
CONFIG_DEVICE_SHELL=y
# CONFIG_DATE_SHELL=y
# CONFIG_CBPRINTF_NANO=y
CONFIG_SENSOR_SHELL=n
CONFIG_SHELL_CMDS_SELECT=y
```

For me, it was necessary to add 

```
CONFIG_MDIO=y
CONFIG_MDIO_NXP_MCUX=y
```

Result:
```
teensy:~$ net iface 1

Interface 0x20202d18 (Ethernet) [1]
===================================
Link addr : 02:04:9F:B1:23:5A
MTU       : 1500
Flags     : NO_AUTO_START,IPv4
Ethernet capabilities supported:
        Virtual LAN
        Auto negotiation
        10 Mbits
        100 Mbits
IPv4 unicast addresses (max 2):
        192.168.179.67 DHCP preferred
        169.254.43.15 autoconf preferred infinite
IPv4 multicast addresses (max 2):
        <none>
IPv4 gateway : 192.168.179.1
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 864000
DHCPv4 renew time : 432000
DHCPv4 server     : 192.168.179.1
DHCPv4 requested  : 192.168.179.67
DHCPv4 state      : bound
DHCPv4 attempts   : 1
```

dleach02, 2022-02-01:
@EmbedWise can you supply a PR with your changes?

closed 2022-04-17 by github-actions[bot]

Rooholla-KhorramBakht, 2024-07-17:
I've been trying to get ethernet on the teensy 4.1 working with the new nxp_enet_mac driver. What is the right dts and prj.conf settings required for this?