zephyrproject-rtos/zephyr #42187: Settings tests are not correctly run

MathieuGeorges, 2022-01-26:
**Describe the bug**

When I build and run the tests for "system.settings.functional.nvs" for native_posix board, I only obtain this ouput : 
```
*** Booting Zephyr OS build v2.7.99-2256-g8aeed7eaba71  ***
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

It looks like no tests are run.
While debugging this unexpected behaviour, I found that the `test_main` function from `settings_basic_test.c` is not called but it calls the weaked one from `ztest.c` : `void __weak test_main(void)`. That is why nothing is done when I run these tests.
Settings tests seems to be the only ones implementing a zephyr_library which call the ztest framework. It could be the reason why the function `test_main` is not called but I did not dig further.

**To Reproduce**
Steps to reproduce the behavior:
`west build -b native_posix tests/subsys/settings/functional/nvs/ -p -t run`

**Expected behavior**
Correctly run all the tests compiled. For instance, it can generate this output : 

```
*** Booting Zephyr OS build v2.7.99-2256-g8aeed7eaba71  ***
Running test suite settings_test_suite
===================================================================
START - test_clear_settings
 PASS - test_clear_settings in 0.0 seconds
===================================================================
START - test_support_rtn
 PASS - test_support_rtn in 0.0 seconds
===================================================================
START - test_register_and_loading
[00:00:00.000,000] <inf> fs_nvs: 4 Sectors of 4096 bytes

[...]

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

**Impact**
While running the tests with twister, it could consider that tests are all passed but nothing is done.

**Environment (please complete the following information):**
- OS: WSL2 on Windows
- Toolchain Zephyr SDK : 13.1

nvlsianpu, 2022-02-02:
Looks like following commit caused this regression: https://github.com/zephyrproject-rtos/zephyr/commit/27f6a5e07d5afe88ade40570716ae498e97e4736.

nvlsianpu, 2022-02-02:
@MathieuGeorges This issue is caused by the tests-framework regression which I described here: https://github.com/zephyrproject-rtos/zephyr/issues/42396 (I used your description as the base).

carlescufi, 2022-02-08:
This is covered by the more general issue here: #42187

closed 2022-02-08 by carlescufi