zephyrproject-rtos/zephyr #42197: Bluetooth: Controller: llcp: No disconnect if remote does not response for initiated control procedure

ppryga, 2022-01-26:
**Describe the bug**
Two connected devices are required.
First device periodically initiates some control procedure e.g. CTE request.
Second device responses to the control procedure as expected.
At the same time second device starts some other control procedure.
For some reason (even due to bug) the first device never sends response to procedure initiated by second device.

The connection is maintained and PDUs are exchanged.

**Expected behavior**
According to BT 5.3 Core specification Vol 6 Part B section 5.2:
> To be able to detect a non-responsive Link Layer Control procedure, both the
Central and the Peripheral shall use a procedure response timeout timer, TPRT.
Upon the initiation of a procedure, the procedure response timeout timer shall
be reset and started.
Each LL Control PDU that is queued for transmission resets the procedure
response timeout timer.
When the procedure completes, the procedure response timeout timer shall be
stopped.
If the procedure response timeout timer reaches 40 seconds, the ACL
connection is considered lost (see Section 4.5.12). The Link Layer exits the
Connection state and shall transition to the Standby state. The Host shall be
notified of the loss of connection.

It means, in my humble opinion, that the connection should be terminated.

**Additional context**
The issue was found during debugging of some other problem.
It was easier to discover the behavior e.g. if there is not enough contexts to handle remote control procedure by first (non-responsive) device. Then, due to missing queueing of remote control procedures, the remote request is lost and never handled.

kruithofa, 2022-01-27:
I agree that the connection should be terminated

kruithofa, 2022-02-08:
I believe that this issue also exists in the legacy LLCP controller

kruithofa, 2022-02-10:
We should do a code inspection in particular for procedures with an instant

kruithofa, 2022-03-10:
Thomas has been able to reproduce, and working on a fix

kruithofa, 2022-06-16:
Issue is fixed with PR #44275 in the 3.1 release

closed 2022-06-16 by kruithofa