zephyrproject-rtos/zephyr #42207: LOG2 + Systemview leads to kernel panic.

manoj153, 2022-01-27:
When LOG2 + Systemview is set it will lead to kernel panic.
Board: `nucleo_h723zg`

**Impact**
Panic  (showstopper)

**Kconfig**
```
CONFIG_LOG=y
CONFIG_PRINTK=y
CONFIG_HEAP_MEM_POOL_SIZE=8192
CONFIG_ASSERT=y
CONFIG_GPIO=y
CONFIG_UART_ASYNC_API=y
CONFIG_LOG_BACKEND_SHOW_COLOR=y
CONFIG_NEWLIB_LIBC=y
CONFIG_LOG2_MODE_DEFERRED=y
CONFIG_SHELL=y
CONFIG_SHELL_CMDS=y
CONFIG_GPIO_SHELL=y
CONFIG_LOG_BACKEND_UART=y
CONFIG_SHELL_LOG_BACKEND=n
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y
CONFIG_PWM=y
CONFIG_LOG_TIMESTAMP_64BIT=y
CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP=n
CONFIG_SHELL_LOG_BACKEND=n
CONFIG_STM32_BACKUP_SRAM=y
CONFIG_REBOOT=y
CONFIG_LOG_MODE_OVERFLOW=y
CONFIG_LOG_BUFFER_SIZE=4096
CONFIG_LOG_SPEED=y
CONFIG_CBPRINTF_FP_SUPPORT=y
CONFIG_COUNTER=y
CONFIG_COUNTER_RTC_STM32_CLOCK_LSE=y
CONFIG_COUNTER_RTC_STM32_BACKUP_DOMAIN_RESET=n
CONFIG_COUNTER_RTC_STM32_SAVE_VALUE_BETWEEN_RESETS=y
CONFIG_LOG_FUNC_NAME_PREFIX_DBG=n
CONFIG_SHELL_ARGC_MAX=16
CONFIG_NOCACHE_MEMORY=y
CONFIG_TRACING=y
CONFIG_SEGGER_SYSTEMVIEW=y
```

**Logs and console output**
Crash log:
```
[946685754] <err> os: r0/a1:  0x00000004  r1/a2:  0x000000a8  r2/a3:  0x240038a8
[946685754] <err> os: r3/a4:  0x0000000f r12/ip:  0x00000000 r14/lr:  0x0801c113
[946685754] <err> os:  xpsr:  0x61000000
[946685754] <err> os: Faulting instruction address (r15/pc): 0x0803111c
[946685754] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[946685754] <err> os: Current thread: 0x240038a8 (logging)
```

On Zephyr e4da3e528088a34a9989f5a50e7ed3149d57de92

nordic-krch, 2022-02-01:
Something similar got fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/41971
That was the issue: https://github.com/zephyrproject-rtos/zephyr/issues/42164
Are you able to check if your issue is also gone? I will see if i can backport it.

nordic-krch, 2022-02-01:
I created backport PR with fix mentioned above. Can you check it? #42346

manoj153, 2022-02-11:
> @manoj153 fix got merged. Can you try it?

Yes, I can, give me 24hrs.

manoj153, 2022-02-14:
> @manoj153 fix got merged. Can you try it?

I have tested. No, it did not solve. Also, please be noted my problem is also with the log redirection.

> It is an issue because I still plan to use the UART logging backend only. Now without per-say / intent, the logs are going to RTT. Is that not a bug?

Which the PR https://github.com/zephyrproject-rtos/zephyr/pull/42209 trying to solve, suggest what will be the best way to go about this issue?

nordic-krch, 2022-02-14:
I would still like to understand what is this error because presence of additional backend should not result in crash. Also sysview should not conflict with RTT use.

Do you have longer crash log? Is this an assert? Can you collect stack trace?

Have you tried to disable logging for rtt `CONFIG_LOG_BACKEND_RTT=n`.

manoj153, 2022-02-22:
> I would still like to understand what is this error because presence of additional backend should not result in crash. Also sysview should not conflict with RTT use.
>
> Do you have longer crash log? [...]
``` 
[946728134] <err> os: r0/a1:  0x00000004  r1/a2:  0x000000a8  r2/a3:  0x24003838
[946728134] <err> os: r3/a4:  0x00000011 r12/ip:  0x00000000 r14/lr:  0x0801bda1
[946728134] <err> os:  xpsr:  0x61000000
[946728134] <err> os: Faulting instruction address (r15/pc): 0x0802e098
[946728134] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[946728134] <err> os: Current thread: 0x24003838 (logging)
```

When  CONFIG_LOG_BACKEND_RTT is disabled, no crash is observed.

manoj153, 2022-02-22:
Unfortunately, CONFIG_DEBUG_COREDUMP=y is not doing its work otherwise I would be able to debug further.

Feel free to share if you have any ideas.

manoj153, 2022-03-22:
@nordic-krch ping

closed 2022-06-05 by github-actions[bot]