zephyrproject-rtos/zephyr #42214: drivers: uart_nrfx_uarte: uart_rx_disable() and uart_rx_enable() fail after reception is complete

mjaun, 2022-01-27:
**Describe the bug**
Related to the `uart_nrfx_uarte` driver: After a UART reception is complete using the async API it is not possible to enable the reception again. A call to `uart_rx_enable()` fails with `EBUSY` while a call to `uart_rx_disable()` fails with `EFAULT`.

Here is how I think the bug is caused:

* If the reception is complete, an ENDRX interrupt is generated which will invoke the user callback.
* Shortly afterwards, a RXTO interrupt is generated which will release the buffer `data->async->rx_buf`.
* Using `uart_rx_enable()` again now is not possible because the flag `data->async->rx_enabled` hasn't been reset.
* Using `uart_rx_disdable()` is also not possible because it checks for a valid buffer `data->async->rx_buf`, but it is currently the only way to reset the flag `data->async->rx_enabled`.

**Expected behavior**
Not sure if one should have to call `uart_rx_disable()` after the reception is complete, but in any case it should be possible to start another reception using `uart_rx_enable()`.

**Impact**
Our current workaround is to call `uart_rx_disable()` immediately in the UART callback because then the buffer isn't released yet. Otherwise this would be a blocker.

**To Reproduce**
I created a sample application to demonstrate the bug here:
https://github.com/mjaun/zephyr/tree/nrf_uart_bug/samples/drivers/uart/nrf_uart_bug

The sample runs on the board `nrf52840dk_nrf52840`.
Pin P1.01 must be connected to P1.02.

**Logs and console output**
Output from the provided sample:
```
** Booting Zephyr OS build v3.0.0-rc1-34-gd05fa6626e11  ***
[00:00:00.334,472] <inf> main: data ready:
                               00 01 02 03 04 05 06 07  08 09                   |........ ..      
[00:00:00.335,571] <err> main: failed to disable reception: -14
[00:00:00.335,571] <err> main: failed to enable uart reception: -16
[00:00:00.335,601] <err> main: failed to enable uart reception: -16
```

anangl, 2022-02-03:
@mjaun Indeed, the `rx_enabled` flag was not cleared properly in the `uart_nrfx_uarte` driver. I provided a fix for this in the PR linked above.

> Not sure if one should have to call `uart_rx_disable()` after the reception is complete

No, RX is supposed to be automatically disabled in such case. It is described here: https://github.com/zephyrproject-rtos/zephyr/blob/bd4ddec0c8c822bbdd420bd558b62c1d1a532c16/include/drivers/uart.h#L178-L186

closed 2022-02-05 by nashif

mjaun, 2022-02-08:
Thanks a lot for the feedback and the quick fix!

PR #42455 (2022-02-05): drivers: uart_nrfx_uarte: Fix RX auto disabling routine
This is a follow-up to commit 11bbdb030def3b4a4b97ee380b1091913e61d371.

When RX is automatically disabled because all provided RX buffers have
been filled up, the rx_enabled flag needs to be cleared, otherwise it
will be impossible to enable RX again.

Add also a test case that will make sure that UART drivers properly handle disabling
and reenabling of RX in async API.

Fixes #42214.