zephyrproject-rtos/zephyr #42227: Teensy41 support SDHC - Storage init Error

EmbedWise, 2022-01-27:
**Describe the bug**
Working on the Teensy41 and trying to get the SD Card functionality to work. For this using the application developed by @bdkrae / @ufechner7. See https://github.com/bdkrae/zephyr_teensy4_test. Which is based on the `zephyr/samples/subsys/fs/fat_fs`.

Please also mention any information which could help others to understand
the problem you're facing:
 - Using a Teensy41
 - SanDisk Edge 4GB Micro SD, both tried with a formatted (MS-DOS and ExFAT partition)
 
I have also tried to follow the instructions 

```
Card voltage switch
Teensy 4.1 has no hardware to switch card voltage. As a workaround in the file zephyr/drivers/disk/usdhc.c the line 502 needs to be changed:

#define SDMMCHOST_NOT_SUPPORT USDHC_HOST_CTRL_CAP_VS18(1)

This prevents the driver from trying to switch over to 1,8V supply for the card. Also, a dummy pin is defined in board defs.
```

I cannot find this Macro where that is defined. I already tried using the FLAG directly USDHC_HOST_CTRL_CAP_VS18_MASK. Still the same error.

**To Reproduce**
Using Zephyr 2.7.1

**Expected behavior**
SD Card should have been mounted

**Logs and console output**
```
[00:00:00.501,000] <inf> usdhc: USDHC detection other than GPIO
[00:00:00.501,000] <err> sdhc: Storage init ERROR!
```

**Environment:**
 - OS: MacOS
 - Toolchain GNU  10.3.1
 - Zephyr 2.7.1

EmbedWise, 2022-01-28:
I tested the interface with Arduino and the same SD card. Which works and shows both information and files on the SD card.

EmbedWise, 2022-01-28:
When the function usdhc_board_access_init is ran in zephyr/driver/disk/usdhc.c 

```
if (!priv->detect_gpio) {
		LOG_INF("USDHC detection other than GPIO");
		priv->inserted = true;
		/* DATA3 does not monitor card insertion */
		base->PROT_CTRL &= ~USDHC_PROT_CTRL_D3CD_MASK;
		if ((base->PRES_STATE & USDHC_PRES_STATE_CINST_MASK) != 0) {
			priv->inserted = true;
		} else {
			priv->inserted = false;
			return -ENODEV;
		}
	} else {
		ret = usdhc_cd_gpio_init(priv->detect_gpio,
				config->detect_pin,
				config->detect_flags,
				&priv->detect_cb);
		if (ret) {
			return ret;
		}
		ret = gpio_pin_get(priv->detect_gpio, config->detect_pin);
		if (ret < 0) {
			return ret;
		}

		gpio_level = ret;

		if (gpio_level == 0) {
			priv->inserted = false;
			LOG_ERR("NO SD inserted!");

			return -ENODEV;
		}

		priv->inserted = true;
		LOG_INF("SD inserted!");
	}
```
Additional information:
The card present switch is not connected to the i.mxrt of the Teensy 4.1 . 

So the cd-gpios in the dts should not be there.

The statement `if ((base->PRES_STATE & USDHC_PRES_STATE_CINST_MASK) != 0) { ` evaluates false and therefore does not detect the SD card. I could not find where this flag should be set.

**Result is the card is never detected**

A dirty workaround is to add this to the dts or overlay and then it works:

```
&usdhc1 {
	status = "okay";
	no-1-8-v;
	cd-gpios = < &gpio3 21 GPIO_ACTIVE_HIGH>;
};
```

Note that I also add no-1-8-v; 
Which applicable to the Teensy4.1

jfischer-no, 2022-01-31:
@EmbedWise , would you like to open a PR to fix it in `boards/arm/teensy4/teensy41.dts` please?

danieldegrasse, 2022-02-02:
@EmbedWise I don't have a teensy board to test this fix on, but looking at the schematic for the board it looks like the chip detect line for the SD card is not routed. The USDHC driver supports detecting a card via the internal pull up resistor on the DAT3 line. I've opened a PR (#42379) to switch the `teensy41` board to using the DAT3 method of detecting SD card presence, instead of the dummy GPIO that was in use. Can you test that PR and verify it works for you? I would use the `tests/drivers/disk` test.

danieldegrasse, 2022-02-02:
The PR is in `main` now, so you should be able to verify the fix with the latest version of Zephyr. I tested the fix on an RT1060 EVK, which has the same SoC as the teensy 4.1, but I would still like to get your feedback if possible to ensure the fix was valid @EmbedWise

EmbedWise, 2022-02-08:
@danieldegrasse I will test it and revert back to you.

EmbedWise, 2022-02-09:
> The PR is in `main` now, so you should be able to verify the fix with the latest version of Zephyr. I tested the fix on an RT1060 EVK, which has the same SoC as the teensy 4.1, but I would still like to get your feedback if possible to ensure the fix was valid @EmbedWise

I verified it on the Teensy board but it is not detecting the SD card. Code that I used can be found on Github https://github.com/EmbedWise/zephyr_teensy4_app 

If I change it to a dummy GPIO it works and detects the card.

danieldegrasse, 2022-02-09:
@EmbedWise, taking another look at this, there is no power control circuitry for the SD card, so that may be causing the SD card detection with DAT3 to fail. What specific error do you see when using DAT3 card detection, as opposed to the dummy GPIO?

EmbedWise, 2022-03-01:
@danieldegrasse there is indeed no power control circuit. The error I see it the following:

```
[00:00:02.001,000] <inf> main: Started zephyr 3.0.0 on board arm/teensy41.
[00:00:05.000,000] <inf> usdhc: USDHC detection other than GPIO
[00:00:05.001,000] <inf> usdhc: USDHC detection using DAT3 pull
[00:00:05.001,000] <err> sdhc: Storage init ERROR!
```

bearsh, 2024-08-20:
I just tried the `samples/subsys/usb/mass` example and it still does not work out of the box...

maybe this issue should get reopened.

danieldegrasse, 2024-08-20:
> I just tried the samples/subsys/usb/mass example and it still does not work out of the box...

> maybe this issue should get reopened.

Probably so, I'll go ahead and do that. This board doesn't have any card detect circuitry, so I think the best solution may be to modify the driver to assume a card is present if no card detect method is configured.

danieldegrasse, 2024-08-20:
@bearsh can you try #77301 on your board?

closed 2024-08-22 by nashif

bearsh, 2024-08-26:
> ~@bearsh can you try #77301 on your board?~ or just try the latest `main`, as that PR has merged

yes, works! thanks!

PR #42379 (2022-02-02): boards: teensy41: Enable DAT3 pull detection for USDHC
This PR started as fix for #42227, but when I tested the fix on an RT1060 the test itself also failed with a usage fault, so the PR fixes that too.
Details on the fix for #42227:
`teensy41` has no chip detect signal for SD slot. Use pull up resistor present on SD cards on DAT3 to detect card instead. Remove software defined chip detect IOMUXC selection that was used as a dummy chip detect pin.

Details on the fix for #42380:
The RT series does not set a default PLL PFD fractional divider, and without this configured the USDHC driver will e [... cut]
commit 2355fe86 (2022-02-02): boards: teensy41: Enable DAT3 pull detection for USDHC

teensy41 has no chip detect signal for SD slot. Use DAT3 pull card
detection. Remove software defined chip detect IOMUXC selection that was
used as a dummy pin.

Fixes #42227
PR #77301 (2024-08-22): drivers: sdhc: imx_usdhc: assume card is present if no detection method
The imx USDHC driver previously queried the peripheral's internal card
detect signal to check card presence if no card detect method was
configured. However, some boards do not route the card detect signal and
do not work correctly with the DAT3 detection method supported by this
peripheral. As a fallback, assume the card is present in the slot but
log a warning to the user.

Fixes #42227
commit 17f71e19 (2024-08-22): drivers: sdhc: imx_usdhc: assume card is present if no detection method

The imx USDHC driver previously queried the peripheral's internal card
detect signal to check card presence if no card detect method was
configured. However, some boards do not route the card detect signal and
do not work correctly with the DAT3 detection method supported by this
peripheral. As a fallback, assume the card is present in the slot but
log a warning to the user.

Fixes #42227