zephyrproject-rtos/zephyr #42235: Socket CAN not supported for NUCLEO H743ZI?

bennettbe, 2022-01-27:
Running the sample **zephyr/samples/net/sockets/can/** with build and flash commands,

```
west build -p -b nucleo_h743zi zephyr/samples/net/sockets/can/
west flash
```

fails after flashing and the board outputs.

```
*** Booting Zephyr OS build v3.0.0-rc1-2-g140837ce3ac6  ***

[00:00:02.005,000] <err> net_socket_can_sample: No CANBUS network interface found!
[00:00:02.015,000] <err> net_socket_can_sample: Cannot start CAN application (-2)
uart:~$ 
```

I found a similar issue that was report back in 2020, https://github.com/zephyrproject-rtos/zephyr/issues/30432. Commits were added to fix this issue. 

Is there socket can support for the Nucleo H743zi?

erwango, 2022-01-28:
@alexanderwachter Would you mind having a look ? (at least to see of this isn't a basic configuration issue)

alexanderwachter, 2022-01-28:
#41847 will solve this issue

closed 2022-04-08 by carlescufi