zephyrproject-rtos/zephyr #42273: CONFIG to define a __FILENAME__ and use that instead of __FILE__ to reduce flash usage

waelbarakat, 2022-01-28:
**Is your enhancement proposal related to a problem? Please describe.**

In order to reduce flash usage, we're trying to minimize the strings that are compiled in, mainly from using asserts defined in `__assert.h`.

Many strings show up as
```
WEST_TOPDIR/zephyr/drivers/clock_control/nrf_clock_calibration.c
```

There's a config to help reduce the strings by removing the condition being tested `CONFIG_ASSERT_NO_COND_INFO` but it would be great if there could be one to use the filename instead of the file+path.

**Describe the solution you'd like**

Usually we can pass in `-D__FILENAME__="$(notdir $<)"` to the `CFLAGS` and that would strip the full path and just use the filename only.

**Describe alternatives you've considered**

I tried adding that via a `zephyr_compile_options` but that's leading to build errors.
```
arm-none-eabi-gcc: error: $(notdir $<): No such file or directory
```

Did someone get this to work?

waelbarakat, 2023-03-28:
Hey @henrikbrixandersen, I see this is assigned to you now. Do you have an idea if this is easily doable with existing build infrastructure?

henrikbrixandersen, 2023-03-28:
I am sorry, but this is not assigned to me and I'm afraid I don't have a solution to this.