zephyrproject-rtos/zephyr #42277: Zephyr Docs on West need to be updated to include SBOM generation

kestewart, 2022-01-28:
**Describe the bug**
The project documentation does not describe how to use the west spdx command to build SBOMs

Please also mention any information which could help others to understand
the problem you're facing:
People are interested in generating SBOM's and while Zephyr has the capability, it is not documented in Zephyr's public documents.    To get the information on how to invoke it, it's necessary to go to https://github.com/zephyrproject-rtos/zephyr/commit/fd31b9b4acb5d48f83ef8cf0f3afe3e48b7844c0

**To Reproduce**
Steps to reproduce the behavior:
1. https://docs.zephyrproject.org/latest/guides/west/release-notes.html# - contains no mention of this capability added last May.
2. https://docs.zephyrproject.org/latest/guides/west/built-in.html - contains not mention of the west spdx command

**Expected behavior**
Please add the relevant information on invoking the SBOM creation capability to the project documentation.

**Impact**
This is a feature that is increasingly being used by Zephyr Users, and making it easier for them to understand how create these SBOMs will improve the user experience.  

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - OS: all
 - Toolchain: West
 - Commit SHA or Version used: This capability has been included since 2.6

**Additional context**
N/A

MaureenHelm, 2022-01-29:
`west spdx` is an extension command, not a built-in command:
https://docs.zephyrproject.org/latest/guides/west/zephyr-cmds.html#software-bill-of-materials-west-spdx

kestewart, 2022-01-31:
Thanks Maureen.     That was what I was looking for, but failing to find. 

Was surprised though there was no mention in any of the release notes.

MaureenHelm, 2022-01-31:
It was mentioned in the 2.6 and 2.7 release notes:
```
~/zephyrproject/zephyr$ git grep -i spdx -- doc/releases/
doc/releases/release-notes-1.13.rst:* :github:`7314` - Generate SPDX TagValue document as part of 1.13 release
doc/releases/release-notes-1.14.rst:* :github:`15234` - Missing SPDX license references in drivers source files.
doc/releases/release-notes-1.7.rst:* :jira:`ZEP-1457` - Add SPDX Tags to Zephyr license boilerplate
doc/releases/release-notes-1.7.rst:* :jira:`ZEP-1593` - /scripts/sysgen should create output using SPDX licensing tag
doc/releases/release-notes-2.5.rst:* :github:`32194` - Source files missing specification of SPDX-License-Identifier in comments
doc/releases/release-notes-2.5.rst:* :github:`31938` - Invalid SPDX license identifier used in file
doc/releases/release-notes-2.6.rst:  * ``west spdx``: new command which can be used to generate SPDX software
doc/releases/release-notes-2.6.rst:    bills of materials for a Zephyr application build. See :ref:`west-spdx`.
doc/releases/release-notes-2.6.rst:* :github:`7317` - Add generation of SPDX TagValue documents to each build
doc/releases/release-notes-2.7.rst:* Automatic SPDX SBOM generation has been added to West
```

It's implemented in the main zephyr repo, so you should be looking in the zephyr release notes, not the west release notes.

closed 2022-08-19 by mbolivar-nordic