zephyrproject-rtos/zephyr #42278: USB CDC-ACM non-functional after host reboot

mlsvrts, 2022-01-28:
**Describe the bug**
I'm running a USB CDC-ACM endpoint for both logging and as a uart shell on a NXP RT1050. When I start my application from chip reset, USB works correctly; I can see log messages, and use the uart shell.

After any platform state transition (sleep (S3), hibernation (S4), or off (S5)), the USB CDC-ACM endpoint is no longer functional when the system resumes. It is possible to connect to the virtual COM device, but no log messages are present and the shell will not echo characters. 

**To Reproduce**
1. Boot operating system, reset MCU using external pin
2. Open virtual port in PuTTy, observe log messages
3. Reboot system
4. Open virtual port in PuTTy, observe that log messages are no longer received

**Expected behavior**
The USB device should continue to function after a platform reset.

**Impact**
This is a clear breaking issue; USB is not usable after any power state transition.

**Environment (please complete the following information):**
 - OS: Windows 11
 - SDK 0.13.1
 - Zephyr 2.7.1, West

**Additional context**
I did some searching and found an old issue that seemed to replicate my problem (#15917), and in particular [this gist](https://gist.github.com/acsplayon/b594d6b2207f8923edb071fdee499b36), which fixes the issue by monitoring for USB_DC_RESET events.

Now it has been some time and [forward_status_cb](https://github.com/zephyrproject-rtos/zephyr/blob/ae0297e8de1f94b050de0afecc9e2150006e3ec6/subsys/usb/usb_device.c#L1181) does not check for USB_DC_RESET. However, it does seem to maintain the spirit of the behavior from this old patch, so I have tried a small change to the existing code:

```diff
diff --git a/subsys/usb/usb_device.c b/subsys/usb/usb_device.c
index c96b172b44..de34028eb9 100644
--- a/subsys/usb/usb_device.c
+++ b/subsys/usb/usb_device.c
@@ -1200,10 +1200,10 @@ static void forward_status_cb(enum usb_dc_status_code status, const uint8_t *par
                usb_reset_alt_setting();
        }

-       if (status == USB_DC_DISCONNECTED || status == USB_DC_SUSPEND) {
+       if (status == USB_DC_DISCONNECTED || status == USB_DC_SUSPEND || status == USB_DC_RESET) {
                if (usb_dev.configured) {
                        usb_cancel_transfers();
-                       if (status == USB_DC_DISCONNECTED) {
+                       if (status == USB_DC_DISCONNECTED || status == USB_DC_RESET) {
                                foreach_ep(disable_interface_ep);
                                usb_dev.configured = false;
                        }
```

This change has, in my testing, worked beautifully and fixed all of my issues 🎉. However, I do not have the USB subsystem expertise to know if there is a potential negative impact on other platforms.

jfischer-no, 2022-01-31:
@mlsvrts thanks for report. You changes makes sense at first glance, but there should be USB_DC_DISCONNECTED event, can you please try just the following changes: 

```diff c
diff --git a/drivers/usb/device/usb_dc_mcux.c b/drivers/usb/device/usb_dc_mcux.c
index 0a1bda0f034..205678cf306 100644
--- a/drivers/usb/device/usb_dc_mcux.c
+++ b/drivers/usb/device/usb_dc_mcux.c
@@ -810,6 +810,9 @@ static void usb_mcux_thread_main(void *arg1, void *arg2, void *arg3)
                case kUSB_DeviceNotifyResume:
                        dev_data.status_callback(USB_DC_RESUME, NULL);
                        break;
+               case kUSB_DeviceNotifyDetach:
+                       dev_data.status_callback(USB_DC_DISCONNECTED, NULL);
+                       break;
                default:
                        ep_abs_idx = EP_ABS_IDX(msg.code);
```

mlsvrts, 2022-02-01:
@jfischer-no Thanks for the suggestion, but unfortunately this patch does not fix the issue on its own.

I assume this is because no `kUSB_DeviceNotifyDetach` event ever occurs (or perhaps occurs out of order?), as sending `USB_DC_DISCONNECTED` from the `kUSB_DeviceNotifyBusReset` is sufficient to fix the issue. So:

```diff
case kUSB_DeviceNotifyBusReset:
		handle_bus_reset();
-               dev_data.status_callback(USB_DC_RESET, NULL);
+		dev_data.status_callback(USB_DC_DISCONNECTED, NULL);
		break;
```

Will resolve the issue (functionally, I think this is the same as my previous patch, but limited to MCUX devices). But I think that it may be problematic to send the 'wrong' state event from this case, particularly for anyone refactoring later.

If you have any other suggestions I will happy to help test them!

dkalowsk, 2022-02-01:
Similar to / duplicate of #39426 that was closed due to no response

jfischer-no, 2022-02-01:
> Similar to / duplicate of #39426 that was closed due to no response

No!

jfischer-no, 2022-02-01:
> I assume this is because no `kUSB_DeviceNotifyDetach` event ever occurs (or perhaps occurs out of order?), as sending `USB_DC_DISCONNECTED` from the `kUSB_DeviceNotifyBusReset` is sufficient to fix the issue.

@mlsvrts Probably something is still missing, I will take a closer look, it would be nice to have USB_DC_DISCONNECTED event working in usb_dc_mcux.c driver. That is probably why I could not observe this issue with nRF USBD driver as there would be a USB_DC_DISCONNECTED event.
Your changes regarding USB_DC_RESET seems to be correct, would you like to open a PR?

mlsvrts, 2022-02-02:
> Your changes regarding USB_DC_RESET seems to be correct, would you like to open a PR?

@jfischer-no Thank you for the quick response and review! I have created a PR with my changes. I have not contributed to Zephyr before, so please let me know if I need to fix anything in the request.

> Probably something is still missing, I will take a closer look, it would be nice to have USB_DC_DISCONNECTED event working in usb_dc_mcux.c driver. 

In a few weeks I should have working SWD, so I may further investigate the mcux behavior then; it is challenging to look deeper now, as I have to rely on USB logging to debug itself :).

PR #42383 (2022-02-04): usb: Handle USB_DC_RESET event notifications
This patch fixes an issue where mcux usb devices may stop
functioning after a reset event, because they do not
provide a USB_DC_DISCONNECTED event and USB_DC_RESET
events were not handled by the forward status callback.

Fixes #42278
closed 2022-02-04 by carlescufi