zephyrproject-rtos/zephyr #42279: The pthreads are not working on user space. ARM64 cortex_a53 but generic requirement.

AshwinKrishn, 2022-01-28:
**Describe the bug**
Pthread is not working on ARM user space. The system exits when pthreads start.

Please also mention any information which could help others to understand
the problem you're facing:
 - The pthread requires stack to execute and creating the stack array is not possible in user space. 
 - Is there a work around to provide it stack ?

**To Reproduce**
Steps to reproduce the behavior:
1. enable pthreads with user space 

**Expected behavior**
Pthreads must work when called. 

**Impact**
Unable to run multithreaded applications on arm64.

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain Zephyr SDK

dkalowsk, 2022-02-01:
This is not supported and closing as a bug.  Please re-file as a feature request.

cfriedt, 2022-02-01:
@AshwinKrishn - just thought I would chime in. This has been a feature in development for a long time. It's been on the backburner for me for a while. But yea, the assessment is accurate. There needs to be a new first-class Zephyr object type for thread stacks, which requires dynamic objects. 

See also #26999, #24714, #25973