zephyrproject-rtos/zephyr #42288: Use twister with out of tree drivers.

helo9, 2022-01-29:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to run twister in an application with out of tree drivers. As far as I could see, that is not possible atm.

**Describe the solution you'd like**
Add a option to define an module path (where _zephyr/module.yml_ is living and use the folders given there.

**Describe alternatives you've considered**
Manually set the pathes, maybe?

**Additional context**
-

henrikbrixandersen, 2022-01-29:
Please describe the problem you are seeing. I am using twister with out-of-tree drivers just fine.

helo9, 2022-01-29:
Ah nice, I get this error:
```
-- Found BOARD.dts: /home/jonathanh/Programmieren/controllearning/frickelflightcontrol/boards/arm/openpilot_cc3d/openpilot_cc3d.dts
devicetree error: node '/soc/serial@40013800/srxl2' compatible 'spektrum,srxl2receiver' has unknown vendor prefix 'spektrum'
CMake Error at /home/jonathanh/Programmieren/controllearning/zephyr/cmake/dts.cmake:205 (message):
  gen_defines.py failed with return code: 1
```
when running `../zephyr/scripts/twister -T samples -A boards` in my application. But when looking into my normal builds again, I see the same message, it is just ignored from the build system. So I rather have to see how to make spektrum a known vendor or rename it. Will close and eventually reopen, in case problem is present afterwards.

closed 2022-01-29 by helo9

henrikbrixandersen, 2022-01-29:
The procedure for adding new vendor prefixes, which also applies to out-of-tree bindings, is described here:
https://docs.zephyrproject.org/latest/guides/dts/bindings.html#vendor-prefixes