zephyrproject-rtos/zephyr #42290: ESP32 - Heltec Wifi - Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 26 MHz

snarkipus, 2022-01-29:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/42283

<div type='discussions-op-text'>

<sup>Originally posted by **snarkipus** January 28, 2022</sup>
Hello - I'm trying to build the sample/Hello_World file for the [Heltec Wifi Kit 32](https://heltec.org/project/wifi-kit-32/)

The project builds just fine, but it looks like the config is locked to 40MHz so the monitor output is garbled (described [here](https://esp32.com/viewtopic.php?t=2800) I think).

Anyone else been successful?

Search yielded [THIS](https://github.com/zephyrproject-rtos/zephyr/discussions/36827) which doesn't appear to do anything.

esp32.dtsi:

`		rtc: rtc@3ff48000 {
			compatible = "espressif,esp32-rtc";
			reg = <0x3ff48000 0x0D8>;
			label = "RTC";
			xtal-freq = <ESP32_CLK_XTAL_26M>;
			#clock-cells = <1>;
			status = "ok";`

`(.venv) snarkipus@Zeus:~/zephyrproject/zephyr/samples/hello_world$ west espressif -p /dev/ttyUSB0 monitor
toolchain: xtensa-esp32-elf
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7180
ho 0 tail 12 room 4
load:0x40078000,len:14312
ho 0 tail 12 room 4
load:0x40080400,len:3688
entry 0x40080678
0x40080678: esp_ptr_executable at /home/snarkipus/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/include/soc/soc_memory_layout.h:168

W (60) rtc_clk_init: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 26 MHz.
����ln�
        b�lb��nn�l�l��b�����n�r��b��brl�
                                              lb�p��l��r���n�����nn�nn��br�n�lnl����ln�
                                                                                             b�lb��nn���n�nn`</div>

sylvioalves, 2022-02-02:
@snarkipus Hi, which main branch revision you are working with?

snarkipus, 2022-02-02:
@sylvioalves Hey - I'm now on 3.0.0-rc1. This is most likely due to the default crystal frequency setting of 40MHz vs. the installed 26MHz xtal. I had a similar [issue](https://github.com/apache/incubator-nuttx/issues/5380) when trying to build Nuttx as well which was resolved with a change to the clock configuration logic.

Nuttx.patch
`diff --git a/arch/xtensa/src/esp32/esp32_clockconfig.c b/arch/xtensa/src/esp32/esp32_clockconfig.c
index 27a0a0a9c6..89f30ce901 100644
--- a/arch/xtensa/src/esp32/esp32_clockconfig.c
+++ b/arch/xtensa/src/esp32/esp32_clockconfig.c
@@ -25,6 +25,8 @@
 #include <nuttx/config.h>
 #include <stdint.h>

+#include <arch/board/board.h>
+
 #include "xtensa.h"
 #include "xtensa_attr.h"
 #include "hardware/esp32_dport.h"
@@ -186,7 +188,7 @@ void esp32_clockconfig(void)
   uint32_t freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ;
   uint32_t old_freq_mhz;
   uint32_t source_freq_mhz;
-  enum esp32_rtc_xtal_freq_e xtal_freq = RTC_XTAL_FREQ_40M;
+  int xtal_freq = BOARD_XTAL_FREQUENCY / MHZ;

   old_freq_mhz = esp_rtc_clk_get_cpu_freq();
   if (old_freq_mhz == freq_mhz)`
   
However, in the Nuttx case I built the boodloader from source as to modify the default kernel config value to 26MHz. With Zephyr that option isn't available (that I can find).

Note that it does seem to work (at least Hello World) after a string of garbled symbols once it boots. I don't know if that's what is impacting my i2c init though.

closed 2022-05-22 by github-actions[bot]