zephyrproject-rtos/zephyr #42292: Compilation failed: Driver MPU6050

muhady, 2022-01-29:
When I compile Zephyr project with MPU6050 sensor I got an error:

.....
[64/161] Building C object modules/nrf/lib/fatal_error/CMakeFiles/..__nrf__lib__fatal_error.dir/fatal_error.c.obj
[65/161] Building C object modules/nrf/drivers/hw_cc310/CMakeFiles/..__nrf__drivers__hw_cc310.dir/hw_cc310.c.obj
[66/161] Linking C static library zephyr\drivers\i2c\libdrivers__i2c.a
[67/161] Building C object zephyr/drivers/sensor/mpu6050/CMakeFiles/drivers__sensor__mpu6050.dir/mpu6050.c.obj
FAILED: zephyr/drivers/sensor/mpu6050/CMakeFiles/drivers__sensor__mpu6050.dir/mpu6050.c.obj 
C:\nordicsemi\v1.8.0\toolchain\opt\bin\arm-none-eabi-gcc.exe -DBUILD_VERSION=v2.7.0-ncs1 -DKERNEL -DNRF52840_XXAA -DUSE_PARTITION_MANAGER=0 -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -IC:/nordicsemi/v1.8.0/zephyr/include -Izephyr/include/generated -IC:/nordicsemi/v1.8.0/zephyr/soc/arm/nordic_nrf/nrf52 -IC:/nordicsemi/v1.8.0/nrf/include -IC:/nordicsemi/v1.8.0/modules/hal/cmsis/CMSIS/Core/Include -IC:/nordicsemi/v1.8.0/modules/hal/nordic/nrfx -IC:/nordicsemi/v1.8.0/modules/hal/nordic/nrfx/drivers/include -IC:/nordicsemi/v1.8.0/modules/hal/nordic/nrfx/mdk -IC:/nordicsemi/v1.8.0/zephyr/modules/hal_nordic/nrfx/. -IC:/nordicsemi/v1.8.0/modules/debug/segger/SEGGER -IC:/nordicsemi/v1.8.0/modules/debug/segger/Config -IC:/nordicsemi/v1.8.0/zephyr/modules/segger/. -isystem C:/nordicsemi/v1.8.0/zephyr/lib/libc/minimal/include -isystem c:/nordicsemi/v1.8.0/toolchain/opt/bin/../lib/gcc/arm-none-eabi/9.2.1/include -isystem c:/nordicsemi/v1.8.0/toolchain/opt/bin/../lib/gcc/arm-none-eabi/9.2.1/include-fixed -isystem C:/nordicsemi/v1.8.0/nrfxlib/crypto/nrf_cc310_platform/include -Os -imacros C:/development/nordicsemiapps/zephyr_mpu6050_ncs171b/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros C:/nordicsemi/v1.8.0/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/development/nordicsemiapps/zephyr_mpu6050_ncs171b=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/nordicsemi/v1.8.0/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/nordicsemi/v1.8.0=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/drivers/sensor/mpu6050/CMakeFiles/drivers__sensor__mpu6050.dir/mpu6050.c.obj -MF zephyr\drivers\sensor\mpu6050\CMakeFiles\drivers__sensor__mpu6050.dir\mpu6050.c.obj.d -o zephyr/drivers/sensor/mpu6050/CMakeFiles/drivers__sensor__mpu6050.dir/mpu6050.c.obj -c C:/nordicsemi/v1.8.0/zephyr/drivers/sensor/mpu6050/mpu6050.c
In file included from C:\nordicsemi\v1.8.0\zephyr\include\arch\arm\aarch32\arch.h:20,
                 from C:\nordicsemi\v1.8.0\zephyr\include\arch\cpu.h:19,
                 from C:\nordicsemi\v1.8.0\zephyr\include\kernel_includes.h:33,
                 from C:\nordicsemi\v1.8.0\zephyr\include\kernel.h:17,
                 from C:\nordicsemi\v1.8.0\zephyr\include\init.h:11,
                 from C:\nordicsemi\v1.8.0\zephyr\include\device.h:29,
                 from C:\nordicsemi\v1.8.0\zephyr\include\drivers\i2c.h:23,
                 from C:\nordicsemi\v1.8.0\zephyr\drivers\sensor\mpu6050\mpu6050.c:9:
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:305:40: error: 'DT_N_INST_0_invensense_mpu6050_BUS_P_label' undeclared here (not in a function)
  305 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                                        ^~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2641:24: note: in definition of macro 'DT_CAT'
 2641 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2029:31: note: in expansion of macro 'DT_PROP'
 2029 | #define DT_BUS_LABEL(node_id) DT_PROP(DT_BUS(node_id), label)
      |                               ^~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2022:25: note: in expansion of macro 'DT_CAT'
 2022 | #define DT_BUS(node_id) DT_CAT(node_id, _BUS)
      |                         ^~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2029:39: note: in expansion of macro 'DT_BUS'
 2029 | #define DT_BUS_LABEL(node_id) DT_PROP(DT_BUS(node_id), label)
      |                                       ^~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2390:33: note: in expansion of macro 'DT_BUS_LABEL'
 2390 | #define DT_INST_BUS_LABEL(inst) DT_BUS_LABEL(DT_DRV_INST(inst))
      |                                 ^~~~~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\sys\util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
   84 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:305:31: note: in expansion of macro 'UTIL_CAT'
  305 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2073:27: note: in expansion of macro 'DT_INST'
 2073 | #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
      |                           ^~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2390:46: note: in expansion of macro 'DT_DRV_INST'
 2390 | #define DT_INST_BUS_LABEL(inst) DT_BUS_LABEL(DT_DRV_INST(inst))
      |                                              ^~~~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\drivers\sensor\mpu6050\mpu6050.c:231:15: note: in expansion of macro 'DT_INST_BUS_LABEL'
  231 |  .i2c_label = DT_INST_BUS_LABEL(0),
      |               ^~~~~~~~~~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:305:40: error: 'DT_N_INST_0_invensense_mpu6050_REG_IDX_0_VAL_ADDRESS' undeclared here (not in a function); did you mean 'DT_N_S_cpus_S_cpu_0_REG_IDX_0_VAL_ADDRESS'?
  305 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                                        ^~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2641:24: note: in definition of macro 'DT_CAT'
 2641 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2300:44: note: in expansion of macro 'DT_REG_ADDR_BY_IDX'
 2300 | #define DT_INST_REG_ADDR_BY_IDX(inst, idx) DT_REG_ADDR_BY_IDX(DT_DRV_INST(inst), idx)
      |                                            ^~~~~~~~~~~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\sys\util_internal.h:84:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
   84 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:305:31: note: in expansion of macro 'UTIL_CAT'
  305 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2073:27: note: in expansion of macro 'DT_INST'
 2073 | #define DT_DRV_INST(inst) DT_INST(inst, DT_DRV_COMPAT)
      |                           ^~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2300:63: note: in expansion of macro 'DT_DRV_INST'
 2300 | #define DT_INST_REG_ADDR_BY_IDX(inst, idx) DT_REG_ADDR_BY_IDX(DT_DRV_INST(inst), idx)
      |                                                               ^~~~~~~~~~~
C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:2334:32: note: in expansion of macro 'DT_INST_REG_ADDR_BY_IDX'
 2334 | #define DT_INST_REG_ADDR(inst) DT_INST_REG_ADDR_BY_IDX(inst, 0)
......

****************************************************************************************************
My target platform is NRF52840. When I tried to compile another sample with another sensor e.g. BMM150 I receive very similar error.
Here is Zephyr version: 
-- Zephyr version: 2.7.0 (C:/nordicsemi/v1.8.0/zephyr), build: v2.7.0-ncs1
-- Found Python3: C:/nordicsemi/v1.8.0/toolchain/opt/bin/python.exe (found suitable exact version "3.8.2") found components: Interpreter 
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
*********************************************************************************************************

helo9, 2022-01-29:
That output `C:\nordicsemi\v1.8.0\zephyr\include\devicetree.h:305:40: error: 'DT_N_INST_0_invensense_mpu6050_BUS_P_label' undeclared here (not in a function)` looks like your sensor is missing in [devicetree](https://docs.zephyrproject.org/latest/guides/dts/intro.html#aliases-and-chosen-nodes). In case you didn't add it to an [overlay](https://docs.zephyrproject.org/latest/guides/dts/howtos.html#use-devicetree-overlays), that might be the reason for your problem. The same holds true for the other sensor.

muhady, 2022-01-30:
I didn't add overlay as you mentioned. 
Thank you so much

closed 2022-01-30 by muhady