zephyrproject-rtos/zephyr #42293: Sample: Bluetooth: discovery more than 2 services and its chrc

ck-telecom, 2022-01-30:
**Is your enhancement proposal related to a problem? Please describe.**
I never found an example on dicovery more than 2 services and its chrc, I wonder how it is the best way?

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

jori-nordic, 2024-02-12:
I don't really understand what exactly you want from this ticket.
In any case, there are tests that discover multiple chars under one or more services from the same discovery procedure. That might give you some ideas:

Multiple characteristics from different services:
This one only tests the characteristics' UUIDs, since they are not duplicated in the DB.
https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bsim/bluetooth/host/gatt/settings/src/gatt_utils.c#L144

Multiple characteristics from the same service:
https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bsim/bluetooth/host/gatt/authorization/src/gatt_client_test.c#L122

There is also a test library that presents a somewhat easier API. You can also use it in your application.
Here's a test that uses it to discover one char. You can call it multiple times until you discovered everything.

https://github.com/zephyrproject-rtos/zephyr/blob/592a7cd6a750bdc715f07f004a25226618a89012/tests/bsim/bluetooth/host/att/pipeline/dut/src/main.c#L256-L276

If you want to do it manually, you would just run through a simple state machine in your `discover_func`:
1. discover with type `BT_GATT_DISCOVER_PRIMARY`
2. compare UUID with a list of desired service UUIDs. Once one matches:
3. discover with type `BT_GATT_DISCOVER_CHARACTERISTIC`
4. compare UUID with list of desired char UUIDs (that you expect from this SVC). Once one matches:
5. store char handle
6. goto 1

@Thalley do you discover multiple services at the same time somewhere in the audio stack?

Thalley, 2024-02-12:
> @Thalley do you discover multiple services at the same time somewhere in the audio stack?

Yes the TBS Client (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/audio/tbs_client.c) does that.

However what I think the author here wants may be more related to what the `gatt discover` shell command does (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/shell/gatt.c#L230-L286), i.e. doing discovery with `BT_GATT_DISCOVER_ATTRIBUTE` as the type so that it discovers everything, and the caller can then handle all discovery results in a single callback function (in the case of the shell, it simply prints the handles and UUIDs).