zephyrproject-rtos/zephyr #42308: pm: Force shutdown has no effect

erwango, 2022-01-31:
**Describe the bug**
Running `samples/boards/stm32/power_mgmt/stm32wb_ble`, we can observe that call to `pm_power_state_force((struct pm_state_info){PM_STATE_SOFT_OFF, 0, 0});` has no effect anymore.

When adding following lines after the shutdown command, we can observe the ERROR statement is reached
```
+	printk("ERROR: System off failed\n");
+	while (true) {
+		/* spin to avoid fall-off behavior */
+	}
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b nucleo_wb55rg boards/stm32/power_mgmt/stm32wb_ble
2. west flash

**Expected behavior**
Shutdown should be triggered

**Impact**
Not possible to force shutdown anymore

erwango, 2022-01-31:
Short analysis points to https://github.com/zephyrproject-rtos/zephyr/pull/40339.

r2r0, 2022-01-31:
This is related to issue #41911.
Effectively current implementation of `pm_power_state_force` do nothing.
Sadly #41911 issue has low priority but fortunately fix is pending.

erwango, 2022-01-31:
Closing in favor of https://github.com/zephyrproject-rtos/zephyr/issues/41911

closed 2022-01-31 by erwango