zephyrproject-rtos/zephyr #42323: lwm2m_engine: Error when enabling debug log because of uninitialized variable `from_addr`

petterssonandreas, 2022-01-31:
**Describe the bug**
There is an uninitialized variable `from_addr` in function `socket_recv_message` in lwm2m_engine.c, which is the cause of the following error:
```
[00:07:53.415,954] <err> net_lwm2m_engine: Unknown IP address family:0
[00:07:53.416,015] <dbg> net_lwm2m_engine: lwm2m_udp_receive: checking for reply from [unk]
```

This error only occurs when debug logs are enabled, since the error is actually produced in a function called as an argument to a `LOG_DBG`.

The code in question:
https://github.com/zephyrproject-rtos/zephyr/blob/e83b0741285416fe356eb2d5d0002dddf56a893f/subsys/net/lib/lwm2m/lwm2m_engine.c#L4530-L4562

`from_addr` is never given a value, which will most likely mean that `from_addr.sa_family` will be 0, meaning "Unknown IP address family". Unless I have misunderstood what `recvfrom` does, and if that could write to `from_addr`?

**To Reproduce**
Enable LWM2M logs on DBG level, by setting `CONFIG_LWM2M_LOG_LEVEL_DBG=y` in your `prj.conf`.
Run some code using the `lwm2m_rd_client`, then the error should pop up in the logs.

**Expected behavior**
No error produced. The variable should be initialized.

One way to achieve a correct address is in my case to add:

```
memcpy(&from_addr, &client_ctx->remote_addr, sizeof(from_addr));
```

Is this a good solution? (I'm not sure if remote_addr is always populated)

**Impact**
Only annoying, no impact on functionality.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: gnuarmemb
 - Tested on `v2.7.0-ncs1` but found no fix on latest main either

rlubos, 2022-02-01:
Hi @petterssonandreas,

Since you're referring to NCS, I'm assuming you're using nRF91 modem? There is a known issue, that the offloaded `recvfrom()` does not set the address field when using DTLS.

In general, `recvfrom()` should fill the address buffers when it reports new data, there's no need for the address buffers to be initialized. The proposed solution is more of a workaround for a faulty underlying `recvfrom()` implmentation, I'd rather avoid that in the LwM2M code.

As the nRF91 socket implementation is not a part of Zephyr, but NCS only, it's not really a Zephyr issue - please file a [devzone](https://devzone.nordicsemi.com/) ticket instead.

petterssonandreas, 2022-02-01:
Hi @rlubos,

Yes, I'm using the nRF91 modem. Oh, okay, then I understand. I am not too familiar with sockets so was hard for me to find the correct documentation for it.

But, yes, in that case I agree that my "solution" is incorrect. I will post a ticket at Devzone instead.

Thanks for the input! I will close this issue.

closed 2022-02-01 by petterssonandreas

petterssonandreas, 2022-02-01:
Just as an FYI if anyone else finds this issue, I have opened a ticket on Devzone that you can check out if you are interested: https://devzone.nordicsemi.com/f/nordic-q-a/84271/socket-recvfrom-doesn-t-fill-out-address-buffer