zephyrproject-rtos/zephyr #42325: mgmt/mcumgr/lib:  RFS: Connection/SMP-mcumgr parameters negotiation

de-nordic, 2022-01-31:
**Is your feature request related to a problem? Please describe.**
There are many parameters that affect Connection or mcumgr SMP behavior like buffer size, count, MTU (tx/rx), flash page size, flash alignment, and so on that are currently hardcoded on device side and there is no way for client software to get information on them except for knowing it in advance, for example because vendor know how application for device have been compiled. Some of above parameters should not also be possible to query but also to set and tune, which means that there should be ranges returned for them and a method to set them back.

**Describe the solution you'd like**
The solution needs to take into account that it is not possible, or not really practical, to exchange parameters or set them via CBOR payload as such exchange, when there are a lot of negotiable parameters, may require more CBOR buffer than a device allows.

**Describe alternatives you've considered**
No alternative known yet.

**Additional context**
TBD

de-nordic, 2022-01-31:
FYI @MarekPieta

de-nordic, 2022-02-07:
FYI @carlescufi @philips77

ahedin25, 2022-03-21:
Do you envision these commands being part of a new group or an existing group?

de-nordic, 2022-03-22:
Parameters for groups like fs, image management and so on, should be attached groups.
A problem would be with MTU which are general.

de-nordic, 2023-06-07:
Done with https://github.com/zephyrproject-rtos/zephyr/pull/44643

closed 2023-06-07 by de-nordic