zephyrproject-rtos/zephyr #42327: question: increased power consumption after updating from zephyr 2.4 to zephyr 2.7 on nrf52840

vleeb, 2022-01-31:
hi all,
i just encountered a substantial increase in the overall power consumption of a nrf52840 system in idle-operating mode after updating from zephyr 2.4.x to 2.7.x.
the idle power consumption was about 120uA with zephyr 2.4 and is now about 550uA with zephyr 2.7 ... after doing a direct migration, without any functional changes.
can anybody give me some advise, how to find, which software component(s) might cause this increase? ... or what i can do to reduce the power consumption to the magnitude of 120uA again?
many thanks in advance!
regards,
volker

gmarull, 2022-02-03:
Would you mind sharing some more details, for example:

- Which peripherals are enabled
- Configuration options that your application enables

vleeb, 2022-02-03:
Hi Gerard,

thank you for your reply. 
I am currently digging into the details of this problem and there are some indications that it might have been caused by removing an user-modified (hacked) quectel-modem-driver.
I did not expect that removing a driver will increase the power consumption, but that driver did also some other special hardware configuration settings, which got lost, when removing it.

I am currently removing that driver in the zephyr 2.4 based version and trying to isolate the special hardware configuration settings. 

When I am done with that, I will try to upgrade to 2.7 again. Then I will update this question again.

gmarull, 2022-02-11:
does any of the enabled devices implement support for power management?

vleeb, 2022-02-11:
actively used is the power-management for the uart-driver, because the modem is powerd only two times a day.
bluetooth and inertial module are permanently enabled and working.
spi-nor-flash driver ... is mainly used 2 times a day, but it seems that there is no power-management used.

gmarull, 2022-02-15:
Can you check if the modem driver takes care of suspending the UART device when it no longer needs it? The APIs to do that changed since 2.4, so you may need to double-check this area.

vleeb, 2022-02-15:
Hi Gerard,

thank you for this suggestion. I checked that already, because there was an obvious change in the api. There must be something else …

gmarull, 2022-02-15:
I suggest you do the following:

1. Try to reproduce on a dev kit so we can investigate further
2. Use `git bisect` in the Zephyr repository to find the _culprit_ commit (probably create a minimum application where you can reproduce the behavior before doing this).

vleeb, 2022-02-16:
Hi Gerard,
thanks for supporting me! Git bisect seems to be a good idea.

I will give some updates here, as soon as I can work again on this topic.

closed 2022-05-22 by github-actions[bot]

MarcianoPreciado, 2022-07-27:
@vleeb did you ever find a solution? I'm currently using zephyr 2.7.99 and I'm running into this issue.

vleeb, 2022-07-27:
It seems that the increased power consumption is caused by the uart driver. In the meantime I am trying to upgrade to zephyr 3.2, but I would assume that the problem is still the same.

In my hardware/software setup there seems to be some kind of initialization race-condition, which leads to permanently increased power consumption, when using the interrupt-driven uart api. I was able to avoid this race-condition by using the debugger with an breakpoint somewhere in the init code, but I was not able to find/fix the root cause.

I was able to reduce the power consumption to the expected low value by using the async-dma uarte api, but with that I am currently running into another problem with delayed receiving, which happens only when CONFIG_NETWORKING is enabled.