zephyrproject-rtos/zephyr #42356: Repo size - board documentation - large PNGs

bipson, 2022-02-01:
A ZIP of the current repo is about 80 MB. This is painful for everyone new to the project, especially those not fortunate enough to have a decent Internet connection, but further for every code drop, fork, etc. Even worse it won't ever go away (files are kept in history and will be checked out, even if removed).

This is mostly attributable to images in the 'documentation' for the boards. Sometimes entirely pointless images, like boards/arm/blackpill_f411ce/doc/img/blackpill-v2.jpg, and this practice is questionable in general: Why can't we expect developers or users to look this up from the original sources, i.e. the manufacturer? (Avoiding the whole 'storing PNGs in a git-repo is rather pointless'-discussion for now).
It is rather wasteful to need to push around and store 64 MBs of images that will most certainly almost never be used to write code or deploy to devices.

Some of these images easily approach 1 MB, most have hundreds of KB for very simple content, and considering that there are and will be quite a lot of boards, the repository will certainly grow further.

I think it is too late to remove all the offending images (rewriting history of such a popular repository would be rather painful I fear), I think there should be at least safeguards implemented to disallow merging more image files, or if really necessary, at least only if they are below a certain size (50kB?) -- preferably they should be vector drawings anyway?

I couldn't find anything in the guidelines, so please consider adding this wherever appropriate and sensitize the maintainers for this issue.

carlescufi, 2022-02-01:
This is a reasonable request, but there is a reason why we carry those images, and it is to make it easier to visually identify the boards that we document as supported by Zephyr. We could redirect to the original websites, but having the images is rather useful for those that browse the supported boards.

bipson, 2022-02-01:
Thanks for the quick response, I was hoping there is a reason behind it.

But then the question remains if the pictures really need to be this large (i.e. size on disk)?

gmarull, 2022-02-01:
We could potentially use `.gitattributes` to set a few `export-ignore` rules (haven't tested). Note that doc/ also contains images. So maybe all docs should be excluded from git archives.

nashif, 2022-02-02:
Few things we can do:
- Go over all images and reduce the size to something that is suitable for web, we have a few that is rather very large for no good reason. This something we did in the past.
- I remember some CI/GH Action that would intercept those and force submitted to reduce the size, this for enforcement
- .gitattributes is another option..

nashif, 2022-02-02:
@carlescufi  any particular reason why this one is marked TSC? what are we asking the TSC?

Thalley, 2022-02-03:
The first thing we could do is to see if any of the images exists as SVG which not only scales much better, but require significant less memory. 

The next thing we could do is to compress the images using something as https://tinypng.com/ (or one of the many alternatives), that can often greatly reduce the size of the images without any noticeable loss (assuming that we have the rights to modify the images). 

A more drastic solution could be to move the documentation into a separate git repository, such that a user can clone Zephyr without the documentation, which is available in the browser in any case. 

That all being said, the 80MB isn't all that much compared to e.g. the SDK, so it is a bit curious that the 80MB is an issue when the 1.1GB for the SDK isn't, but that doesn't mean that we can't improve the size of images in Zephyr in any case.

bipson, 2022-02-23:
In my particular case the problem with the ~80 MB (compressed) is that in order to share just the _code_ (for e.g. a code drop accompanying a scientific publication) one needs to push around 80 MB, while most of it is not at all relevant for the code.
Please note that sending patches is not an option, since this can hardly be considered self-contained.

While 80 MB (compressed) might not seem much to some people, for some unprivileged _it is_ and removing just a few simple things leaves 26 MB. Further, new boards are constantly added, so this will grow until it _is_ a problem for more than just the unprivileged. This is basically bloat.

I personally like the idea of splitting the documentation from the code, at least the image files (if they can be seen separated from the rest), considering that they are not at all relevant for the code itself or related to it.

But I also think this is not possible, since there is relevant documentation. Also, just removing or shrinking or compressing the files will only solve part of the problem, since

1. The files will remain in the git tree and thus the repo itself still be bloated,
2. New pictures are added all the time

This is why I would encourage using _preventive_ measures, e.g. make a decision (I can't make) which results e.g. (in the most simple case) in a contribution guideline for new commits/merges that rejects certain file types or above a certain size. And following that encouraging maintainers to enforce this (can be supported with git rules and hooks)

But any other strategy that avoids that problem altogether is of course preferred.

closed 2022-08-29 by carlescufi